commaai/openpilot #24659: C3 Verizon Compatibility

williefeb19, 2022-05-26:
### Describe the bug

Comma support pointed me to creating a bug report.

I currently have a Verizon data only sim card out of a MiFi device in my C3 and have added the appropriate APN. OP says "Connect Online" then after a while it says there is an error for a moment then switches back to online. If I check connect.comma.ai there will be a green dot next to my C3 but if I try to force it upload logs it tells me the device is offline. If I use the Comma app there is no battery voltage reported.

One thing I have tried is setting up a tether on the C3 and connecting with another device to it's WiFi. I am able to get internet. That leads me to believe I have the correct APN settings.

### Provide a route where the issue occurs

2f37c007683e85ba

### openpilot version

0.8.13-release3

### Additional info

I have been unable to find any information as to what carriers are supported or any other technical info on the C3's cellular connection.

williefeb19, 2022-05-27:
I can provide this route -- 2f37c007683e85ba|2022-05-26--20-24-22--3
It goes from online to error twice.

As far as trying [#24663](https://github.com/commaai/openpilot/pull/24663) Do I need to switch to the master branch?

williefeb19, 2022-05-29:
After moving to the master-ci branch version 0.8.15 I am still having the same issues. Connect goes from online to offline and back here is a drive that is from that branch -- 2f37c007683e85ba|2022-05-28--16-02-05

adeebshihadeh, 2022-06-10:
Sorry, I don't have much time to look into it at the moment. We'd like to support as many carriers as possible since the hardware supports it, but we mainly target comma prime compatibility. If you'd like to debug this yourself, you can send AT commands with `mmcli -m 0 --comand="AT+CSQ"`. The Quectel forum may also be useful. Feel free to drop specific questions here too, and I'll answer if I know anything about it.

williefeb19, 2022-06-10:
I did discover how to send commands to the cellular modem by looking through the source code and everything I have tried seems like the modem is working. I am having a hard time troubleshooting because I have been unable to find the relevant logs. I have asked on discord with no response.

I have the following two questions:
1. Is there a log for the cellular connection itself? if so where is it located.
2. From what you posted above and what I have read else where it seems like athena could be part of the issue. Where would logs for this be located?

If you know of any other logs that might be useful I would appreciate it.

adeebshihadeh, 2022-06-10:
1. All the logs are in the systemd journal. Start with this: `journalctl -u ModemManager -f`
2. The athena logs are in the openpilot rlogs. See tools/ for info about how to read those. Athena won't cause your problems though; it's just the most likely place you'll notice these issues since its status is constantly shown in the sidebar.

I also wrote `selfdrive/hardware/tici/restart_modem.sh` while debugging some other modem stuff. It's pretty useful for trying new things out.

adeebshihadeh, 2023-07-01:
AGNOS has improved carrier auto-configuration, which should hopefully improve compatibility with non-comma prime SIMs.

closed 2023-07-01 by adeebshihadeh