commaai/openpilot #24675: 2019 Niro EV OP Long unable to engage

pyoorkate, 2022-05-28:
### Describe the bug

Turning on the OP Long toggle causes open pilot to immediately fault on start up with the error message: "Cruise Fault: Restart the car to engage". (Restarting the car does not work). The car states "Check Forward Collision Avoidance System".

Checked ACCEnable signal of TCS13 with OP Long enabled: Values are 0.

Confirmed Niro EV is not listed in LEGACY_SAFETY_MODE_CAR (in values.py).

Hardware / Software: Comma 2 running 0.8.13.1-release

### Which car does this affect?

2019 Kia Niro EV

### Provide a route where the issue occurs

e809296d6ef28372|2022-05-28--08-53-24--0

### openpilot version

0.8.13.1-release

sshane, 2022-05-29:
Looks like panda was blocking all of openpilot's messages, and then the radar came back up after 20 seconds. I'll see if I can re-run locally to find out why

sshane, 2022-05-29:
Hyundai longitudinal is not a feature on release branches, you can see this functionality is disabled here: https://github.com/commaai/panda/blob/master/board/safety/safety_hyundai.h#L387

closed 2022-05-29 by sshane