commaai/openpilot #24680: Car went onroad without user interaction

kevinharbin, 2022-05-30:
### Describe the bug

The car went onroad/recording with the car off, parked, and no one around.
There were 29 0 min routes starting at 11:49am then a 48 minute route starting at 11:53 that ended when I got back to the car and unplugged the OBD power.

I don't know how to reproduce the bug.

### Provide a route where the issue occurs

f44aa96ace22f34a|2022-05-29--11-49-34--0

### openpilot version

71dca82be1aadc270796e65a1652e2c0251dba9f        civic22_master

sshane, 2022-05-31:
@vanillagorillaa could it be a harness issue? The line we use for ignition is wrong maybe? Think this may be a custom harness

vanillagorillaa, 2022-05-31:
I doubt it's a harness issue. I believe we sent this user a properly built (not a re pin) harness. Also haven't seen this issue previously.

kevinharbin, 2022-05-31:
Confirmed. I'm using the harness VG sent me.

adeebshihadeh, 2022-06-02:
No need for an issue since this car isn't support yet. I copied your issue details into the PR instead.

closed 2022-06-02 by adeebshihadeh