commaai/openpilot #24735: new max speed & speed limits design

pd0wm, 2022-06-03:
Updated Variant 2 based on today's call – [Figma](https://www.figma.com/file/kPvxp4UcTTU9RLLOY7R2b9/openpilot?node-id=367%3A2058)
- MAX
  - updated layout
  - engaged: updated color on "MAX" word (V1 green seems to fit better)
- Speed Limit
  - US
  - EU - enclosing box - bottom right & left have full corner radius for better visual appeal

**Needs decision:** Going over Speed Limit
- Going +10 over speed limit → Max turns orange 🟠
- Going +20 over speed limit → Max turns red 🔴

+10/+20 limits are good enough, or should be made lower? For instance +5/+15?

------

PREVIEW:

[image]

[image]

_Originally posted by @jozef-simo in https://github.com/commaai/openpilot/issues/24681#issuecomment-1145306572_

ClockeNessMnstr, 2022-06-03:
> @jozef-simo What do you want to display if there is no set speed yet? We used to display N/A, but that doesn't quite fit and never thought it looked very good.

"--" is a familiar option from oem systems with set speed or speed limit indicators that don't hide when not available. 
Not sure how prevalent that is actually but.

KyleGreenlaw, 2022-06-04:
Instead of a hard switch at +10/+20 from white->orange->red, maybe change colour in a soft manner when accelerating/decelerating to those speeds .. I feel like it'll be more subtle and less abrupt.

jozef-simo, 2022-06-04:
Great idea! @KyleGreenlaw 
> Instead of a hard switch at +10/+20 from white->orange->red, maybe change colour in a soft manner when accelerating/decelerating to those speeds .. I feel like it'll be more subtle and less abrupt.

PREVIEW:
[image]

maykonpacheco, 2022-06-07:
[image]

PR #24736 (2022-06-15): UI: new set speed design and add speed limits
resolves #24735
closed 2022-06-15 by pd0wm