commaai/openpilot #24745: TSSP reacts poorly to brake commands when creeping

HaraldSchafer, 2022-06-03:
### Describe the bug

see title

### Which car does this affect?

Toyota 2017 prius and TSSP rav4

### Provide a route where the issue occurs

NA

### openpilot version

master

sshane, 2022-07-01:
I suspect this is due to how we request true acceleration, vs. what the PCM expects. When the stock system comes to a stop, the DSU's requested acceleration stays much lower than openpilot's, and instead of ramping to 0, it stays low and moves to its stopping accel of -0.4 m/s/s.

I also suspect the PCM is assuming you want to coast/accelerate when openpilot's request starts to rise to 0, so we need to handle this somehow cleanly in the carcontroller, then we automatically reduce dependence on proportional for stopping, and it should be much smoother.

ntegan1, 2022-09-02:
some tssp pedal corolla drives on a152dad3c3368320|2022-09-01 with toyota-permit-braking + toyota-stopping-control + yesterday's e2e_long retune [device was running this code](https://github.com/ntegan1/openpilot/tree/588660a82b46d71d1d7e975d5f6dba5b966a4471) performed very smoothly in low speed situations with sng traffic, way better than e2e_long before the retune and even more better than stock op long sng

ntegan1, 2022-11-22:
tssp pedal corolla does much better in sng with:

experimental mode,  
lowered pedal scale (reduces sng oovvershooting from stop + harsh braking),  
lowered stopAccel (mostly because -2 is unnecessary and makes loud abs brake pump brrrrr),  
lowered stoppingDecelRate (prevents braking overshooting and is really smooth).

```diff
diff --git a/selfdrive/car/toyota/carcontroller.py b/selfdrive/car/toyota/carcontroller.py
index 4ec950017159a..42293c07b1493 100644
--- a/selfdrive/car/toyota/carcontroller.py
+++ b/selfdrive/car/toyota/carcontroller.py
@@ -47,7 +47,7 @@ def update(self, CC, CS):
       if self.CP.carFingerprint in (CAR.RAV4, CAR.RAV4H, CAR.HIGHLANDER, CAR.HIGHLANDERH):
         PEDAL_SCALE = interp(CS.out.vEgo, [0.0, MIN_ACC_SPEED, MIN_ACC_SPEED + PEDAL_TRANSITION], [0.15, 0.3, 0.0])
       elif self.CP.carFingerprint in (CAR.COROLLA,):
-        PEDAL_SCALE = interp(CS.out.vEgo, [0.0, MIN_ACC_SPEED, MIN_ACC_SPEED + PEDAL_TRANSITION], [0.3, 0.4, 0.0])
+        PEDAL_SCALE = interp(CS.out.vEgo, [0.0, MIN_ACC_SPEED, MIN_ACC_SPEED + PEDAL_TRANSITION], [0.15, 0.32, 0.0])
       else:
         PEDAL_SCALE = interp(CS.out.vEgo, [0.0, MIN_ACC_SPEED, MIN_ACC_SPEED + PEDAL_TRANSITION], [0.4, 0.5, 0.0])
       # offset for creep and windbrake
diff --git a/selfdrive/car/toyota/interface.py b/selfdrive/car/toyota/interface.py
index 3f4edb36d2d12..809bfefdf13a9 100644
--- a/selfdrive/car/toyota/interface.py
+++ b/selfdrive/car/toyota/interface.py
@@ -230,8 +230,10 @@ def get_params(candidate, fingerprint=gen_empty_fingerprint(), car_fw=[], experi
       tune.kpV = [1.3, 1.0, 0.7]
       tune.kiBP = [0., 5., 12., 20., 27.]
       tune.kiV = [.35, .23, .20, .17, .1]
-      if candidate in TSS2_CAR:
-        ret.stoppingDecelRate = 0.3  # reach stopping target smoothly
+      ret.stoppingDecelRate = 0.59  # reach stopping target smoothly
+      ret.stopAccel = -0.6
+      ret.vEgoStarting = 0.32
+      ret.vEgoStopping = 0.32
     else:
       tune.kpBP = [0., 5., 35.]
       tune.kiBP = [0., 35.]
diff --git a/selfdrive/controls/controlsd.py b/selfdrive/controls/controlsd.py
index a18bec83a885b..58821ca1a72e9 100755
--- a/selfdrive/controls/controlsd.py
+++ b/selfdrive/controls/controlsd.py
@@ -108,6 +108,7 @@ def __init__(self, sm=None, pm=None, can_sock=None, CI=None):
     # set alternative experiences from parameters
     self.disengage_on_accelerator = self.params.get_bool("DisengageOnAccelerator")
     self.CP.alternativeExperience = 0
+    self.CP.alternativeExperience |= ALTERNATIVE_EXPERIENCE.RAISE_LONGITUDINAL_LIMITS_TO_ISO_MAX
     if not self.disengage_on_accelerator:
       self.CP.alternativeExperience |= ALTERNATIVE_EXPERIENCE.DISABLE_DISENGAGE_ON_GAS
 
@@ -580,8 +581,7 @@ def state_control(self, CS):
     CC = car.CarControl.new_message()
     CC.enabled = self.enabled
     # Check which actuators can be enabled
-    CC.latActive = self.active and not CS.steerFaultTemporary and not CS.steerFaultPermanent and \
-                   CS.vEgo > self.CP.minSteerSpeed and not CS.standstill
+    CC.latActive = self.active and not CS.steerFaultTemporary and not CS.steerFaultPermanent
     CC.longActive = self.active and not self.events.any(ET.OVERRIDE_LONGITUDINAL) and self.CP.openpilotLongitudinalControl
 
     actuators = CC.actuators
```