commaai/openpilot #24758: Cruise Fault w/ Openpilot Long Toggle ON

tecandrew, 2022-06-06:
### Describe the bug

cruise fault occurs when OP long toggle is on. 

### Which car does this affect?

Hyundai Elantra Hybrid 2021

### Provide a route where the issue occurs

eb5a50f9e98f50b9|2022-06-03--22-09-19

### openpilot version

0.8.14-release3

### Additional info

[Another user claim they see the same issue](https://discord.com/channels/469524606043160576/524611978208215070/982579451281956905) occurs on Kia Niro EV

adeebshihadeh, 2022-06-06:
The Hyundai long safety isn’t included in release3. You’ll need to run a non release branch, e.g. master or devel for Hyundai long.

closed 2022-06-06 by adeebshihadeh

tecandrew, 2022-06-06:
thanks for the info, I had no idea about this. I believe I ran 0.8.13 w/ OP long for a bit w no fault, so I think OP long would work in this release.