commaai/openpilot #24763: Update GM port to use new gear signal with better granularity

JasonJShuler, 2022-06-07:
GM gear signal ECMPRDNL.PRNDL doesn't report low "L" or manumatic gears.

L mode (aka single pedal mode) changes the behavior of the gas pedal such that when fully released, the vehicle applies braking.
Manumatic mode is present on some large-frame vehicles allowing the driver to specify a max gear.

These two additional gears change the acceleration (and possibly braking) behavior of the vehicle, but with the current signal it is impossible to tell.

The new signal - ECMPRDNL2.PRNDL2 supplies all of the drive modes, including the selected manumatic gear.

adeebshihadeh, 2022-06-07:
In the future, no need to create issues if you have a PR ready soon.

adeebshihadeh, 2022-07-17:
Gonna close this since the PR is open.

closed 2022-07-17 by adeebshihadeh