commaai/openpilot #24772: CMake

jwinarske, 2022-06-07:
I have a WIP CMake implementation: https://github.com/jwinarske/openpilot/tree/jw/cmake

Goals:
* Separate source tree from install/"staged install" tree
* Yocto support (any machine arch)
* Macosx x86_64/arm64 support
* Fedora/Ubuntu support
* Eliminate need for binaries in source tree
* Depend on system packages opposed to third_party directory
* SNPE integration via SNPE_ROOT variable

If there is interest I will work towards creating PRs to all the related repos.  I'm pretty close to completion.

adeebshihadeh, 2022-06-07:
Cool project, but we're pretty happy with SCons.

closed 2022-06-07 by adeebshihadeh

jwinarske, 2022-06-07:
@adeebshihadeh With the existing Scons without serious rework Yocto support wouldn't happen anytime soon.  I figured CMake was the fastest way to support Yocto recipes.  I work for Toyota and work on AGL among other things.  That's a shame.

jwinarske, 2022-06-07:
@adeebshihadeh In addition CMake sits next to Scons.  It does not replace it.

geohot, 2022-06-07:
I weighed in on this as well. We like that scons is Python, and have no plans to support Yocto.

We mainly target Ubuntu, with limited support for OS X and Android.

geohot, 2022-06-08:
Similar to my comment on the Rust Panda firmware. We are a company with a clear mission, "solve self driving cars while delivering shippable intermediaries". We are not held back by our Scons build system in any way I'm aware of.

jwinarske, 2022-06-08:
@geohot I suspect that you know this; Yocto is synonymous with embedded Linux.  The de-facto standard for commercial products that ship on Linux use a Yocto based OS image.  It enforces a number of disciplines that enables a greater return from Engineering NRE; helps to minimize tech debt, reproducible OS builds, platform porting is straight forward, lower cost of ownership, common install patterns, board farm testing support, etc.  Yocto provides a solid foundation on which to solve problems.  It's very much in line with your business goals.  I'm not soliciting for your teams engineering bandwidth, only socializing ideas for open source work.