commaai/openpilot #24805: kia mohave 2022 some new signal

crwusiz, 2022-06-10:
kia mohave 2022 not can-fd but HDA2

d209f8624080a09a|2022-06-10--02-25-31--1

comma official release2 0.8.13.1

untitled | 2:363 | 2254 |  
untitled | 0:363 | 2254 |  
untitled | 2:364 | 2252 |  
untitled | 0:364 | 2253 |  

untitled | 2:398 | 901 |  
untitled | 0:398 | 901 |  
untitled | 0:399 | 901 |  
untitled | 2:399 | 901 |  
untitled | 2:39a | 901 |  
untitled | 0:39a | 901 |  
untitled | 2:39b | 901 |  
untitled | 0:39b | 901 |  
untitled | 0:39c | 901 |  
untitled | 2:39c | 901 |  

untitled | 2:437 | 900 |  
untitled | 0:437 | 901 |  
untitled | 0:438 | 900 |  
untitled | 2:438 | 900 |  
untitled | 2:43b | 901 |  
untitled | 0:43b | 901 |  
untitled | 2:443 | 901 |  
untitled | 0:443 | 902 |  
untitled | 0:444 | 901 |  
untitled | 2:444 | 901 |  

untitled | 0:463 | 450 |  
untitled | 2:463 | 450 |  
untitled | 0:479 | 450 |  
untitled | 2:479 | 450 |  

.............. some other unknown signal

               'carFw': [{'address': 2000,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'\xf1\x87991102J730\xf1\x00HM__ SCC -----      1.00 1.00 99110-2J730         \xf1\xa01.00',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 2004,
                          'ecu': 'eps',
                          'fwVersion': b'\xf1\x8757700-2J800\xf1\x00HM  MDPS R 1.00 1.02 57700-2J800 4HS1R102\xf1\xa01.02',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 2001, 'ecu': 'esp', 'fwVersion': b'\xf1\xa01.01', 'responseAddress': 0, 'subAddress': 0},
                         {'address': 2017,
                          'ecu': 'transmission',
                          'fwVersion': b'\xf1\x8795441-4E906\x00\xf1\x81E06\x00\x00\x00\x00\x00\x00\x00\xf1\x00PSBG2333  E06\x00\x00\x00'
                                       b'\x00\x00\x00\x00THM0D30KS3\x9a!\x0c\xeb\xf1\xa0PSBG',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 2016,
                          'ecu': 'engine',
                          'fwVersion': b'\xf1\x8940HMW00A00DIS201\xf1\x8240HMW00A00DIS201',
                          'responseAddress': 0,
                          'subAddress': 0}],

adeebshihadeh, 2022-07-11:
Do you have this working? How is it using the new signals if it's not CAN-FD? Most of the relevant signals are longer than 8 bytes.

crwusiz, 2022-07-12:
> Do you have this working? How is it using the new signals if it's not CAN-FD? Most of the relevant signals are longer than 8 bytes.

I don't own a car, so it's hard to test it additionally. I left it for you to refer to the extracted cabana at the time. The vehicle does not have can-fd applied and a corner radar has been added to view it as a vehicle between the old model year and can-fd.

closed 2022-08-01 by adeebshihadeh