commaai/openpilot #24808: ui: unable to scroll on 'Regulatory' page

BadOlives, 2022-06-10:
### Describe the bug

When selecting the VIEW button on the Regulatory option in Settings, user is unable to scroll down to see additional information.

### Provide a route where the issue occurs

No route

### openpilot version

0.8.15 master

BadOlives, 2022-06-10:
Referencing issue #24800 as this might go along with this issue.

Achilles308, 2022-06-15:
@pd0wm The scrolling on the terms and conditions page when onboarding works great... you have to scroll all the way down for the accept button to be activated though. Here is the part calling the tc.html page
https://github.com/commaai/openpilot/blob/cb8885cffb313bea258c012c3026461a22bf8135/selfdrive/ui/qt/offroad/onboarding.cc#L91

and the terms and conditions are located here https://github.com/commaai/openpilot/blob/cb8885cffb313bea258c012c3026461a22bf8135/selfdrive/assets/offroad/tc.html#L1

Might help to see a working scroll for the html display... seems to be rarely used.

adeebshihadeh, 2022-07-01:
@deanlee want to take a look at this one?

deanlee, 2022-07-01:
I'll check in it.

closed 2022-07-07 by adeebshihadeh