commaai/openpilot #24830: Hyundai Tucson 2022 CAN-FD Can signal

shinkansan, 2022-06-12:
d4f02ccf4a8502f0|2022-06-12--12-30-23

Some IDs are similar to kia ev6, but slightly different. Tuscon is can-fd, but even without side radar, it is classified as HDA2 by the attribute value of the hyundai official maintenance device.

BO_ 426 CRUISE_BUTTONS: 16 XXX
 SG_ ADAPTIVE_CRUISE_BTN : 34|1@0+ (1,0) [0|1] "" XXX
 SG_ LKAS_BTN : 39|1@0+ (1,0) [0|1] "" XXX
 SG_ PAUSE_RESUME_BTN : 38|1@0+ (1,0) [0|1] "" XXX
 SG_ DISTANCE_BTN : 37|2@0+ (1,0) [0|3] "" XXX
 SG_ _CHECKSUM : 7|16@0+ (1,0) [0|65535] "" XXX
 SG_ _COUNTER : 23|8@0+ (1,0) [0|255] "" XXX

BO_ 346 ACCELERATOR: 8 XXX
 SG_ NEW_SIGNAL_3 : 16|8@1+ (1,0) [0|255] "" XXX
 SG_ NEW_SIGNAL_4 : 7|16@0+ (1,0) [0|65535] "" XXX
 SG_ NEW_SIGNAL_2 : 40|3@1+ (1,0) [0|7] "" XXX
 SG_ NEW_SIGNAL_5 : 47|8@0+ (1,0) [0|255] "" XXX
 SG_ onlyworkonP : 56|1@0+ (1,0) [0|1] "" XXX
 SG_ GEAR : 24|3@1+ (1,0) [0|7] "" XXX
 SG_ ACCELERATOR_PEDAL : 55|8@0+ (1,0) [0|255] "" XXX

shinkansan, 2022-06-12:
I confirmed that the vehicle was in control for a while( < 2s) with LFA Signal from ev6 dbc . Afterwards, the same error occurred when the HDA camera harness was disconnected.

What I'm wondering is, if the handle moved the signal sent by OpenPilot for a while, it's not a checksum issue, right? Or is it just made to move the steer for a few frames even if the checksum is wrong?

adeebshihadeh, 2022-07-11:
@sshane let's get this model supported for 0.8.16

@shinkansan can you upload and preserve a route with the stock system enabled?

shinkansan, 2022-07-24:
LFA + AEB Warning - d4f02ccf4a8502f0|2022-07-22--13-54-11

shinkansan, 2022-07-25:
Stock LFA + SCC - d4f02ccf4a8502f0|2022-07-25--12-31-47

sshane, 2022-11-09:
@shinkansan going to work on this port. Are you ready to help test?

sshane, 2022-11-09:
@shinkansan Can you put your device on WiFi and upload a route with rlogs? You can just let me know when it's on WiFi and I can do the rest

sshane, 2022-11-09:
@shinkansan can you test this branch? https://github.com/commaai/openpilot/pull/26427/files

adeebshihadeh, 2022-11-10:
Gonna close this since we have the PR up.

closed 2022-11-10 by adeebshihadeh