commaai/openpilot #24883: Chyrlser: resume spam

adeebshihadeh, 2022-06-16:


sshane, 2022-06-17:
Spamming both busses should work according to @realfast

realfast, 2022-06-17:
I am sure there is a better way

I did the following in carcontroller:

```if CC.cruiseControl.cancel:
      can_sends.append(create_wheel_buttons(self.packer, CS.button_counter + 1, 0, cancel=True, acc_resume = False))
      can_sends.append(create_wheel_buttons(self.packer, CS.button_counter + 1, 2, cancel=True, acc_resume = False))
    elif CS.out.cruiseState.standstill:
      can_sends.append(create_wheel_buttons(self.packer, CS.button_counter + 1, 0, cancel=False, acc_resume = True))
      can_sends.append(create_wheel_buttons(self.packer, CS.button_counter + 1, 2, cancel=False, acc_resume = True))```
      
      And this is chryslercan
      
      ```def create_wheel_buttons(packer, frame, bus, cancel = False, acc_resume = False):
  # Cruise_Control_Buttons Message sent to cancel ACC.
  values = {
    "ACC_Cancel": cancel,
    "COUNTER": frame % 0x10,
    "ACC_Resume": acc_resume,
  }
  return packer.make_can_msg("Cruise_Control_Buttons", bus, values)```

closed 2022-07-07 by adeebshihadeh