commaai/openpilot #24900: 2022 Elantra HEV no longer recognized as supported car

maxtriano, 2022-06-18:
### Describe the bug

Seeing this issue as of yesterday, running master-ci. Openpilot had previously worked, but now shows the car as Unsupported and goes into dashcam mode. carFw shows new fwVersion values for engine and transmission, and openpilot works when these are added to values.py. To my knowledge HKG hasn't pushed out an OTA firmware update, but apparently this is happening for other Hyundai owners with different cars, so it's possible. 

Here's a route I drove with the new fingerprint fwVersions added and openpilot working: 786fc028c014be71|2022-06-18--11-37-25

### Which car does this affect?

2022 Hyundai Elantra HEV

### Provide a route where the issue occurs

786fc028c014be71|2022-06-17--22-18-06

### openpilot version

30c3bc07874cbbc57780e777c156f90d710b54bd

dermida, 2022-06-19:
Afects Hyundai Ioniq EV 2020 to.
Drive  with car unrecognized:
b1078edf825d2db6|2022-06-19--10-41-40

Drive where the car is recognized:
b1078edf825d2db6|2022-06-18--22-30-50

dermida, 2022-06-19:
Examples where car unrecognized and the day before where is not recognized
I changed the USB cable to a new one to

'carFingerprint': 'mock',
'carFw': [{'address': 2004,
                          'ecu': 'eps',
                          'fwVersion': b'\xf1\x00AE  MDPS C 1.00 1.01 56310/G7310 4APEC101',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 2000,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'\xf1\x00AEev SCC F-CUP      1.00 1.01 99110-G7000         ',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 1988,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'\xf1\x00AEE MFC  AT EUR LHD 1.00 1.00 95740-G2600 190730',
                          'responseAddress': 0,
                          'subAddress': 0}],
               'carName': 'hyundai',

'carFingerprint': 'HYUNDAI IONIQ ELECTRIC 2020',
carFw': [{'address': 2004,
                          'ecu': 'eps',
                          'fwVersion': b'\xf1\x00AE  MDPS C 1.00 1.01 56310/G7310 4APEC101',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 2000,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'\xf1\x00AEev SCC F-CUP      1.00 1.01 99110-G7000         ',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 1988,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'\xf1\x00AEE MFC  AT EUR LHD 1.00 1.00 95740-G2600 190730',
                          'responseAddress': 0,
                          'subAddress': 0}],
               'carName': 'hyundai',

haram-KONA, 2022-06-20:
Hyundai KONA Hybrid 2020 is same issues (engine & transmission fw-version changed)

With two C3 devices, each is checked as follows.
(It was checked on the same day, at the same time, in the same vehicle, with the same cable and other conditions.)

1. 0.8.14 release3 is nomal good states.
    0.8.14 release 3 normal good carfw route id : 0665dbbc79ef999e|2022-06-18--15-50-17

log: 
carParams
{'carParams': {'alternativeExperience': 1,
               'carFingerprint': 'HYUNDAI KONA HYBRID 2020',
               'carFw': [{'address': 2017,
                          'ecu': 'transmission',
                          'fwVersion': b'\xf1\x816U3J9051\x00\x00\xf1\x006U3H1_C2\x00\x006U3J9051\x00\x00HOS0G16DS1\x16\xc7\xb0\xd9',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 2016,
                          'ecu': 'engine',
                          'fwVersion': b'\xf1\x816H6F6051\x00\x00\x00\x00\x00\x00\x00\x00',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 1988,

2. 0.8.15 master-ci is carfw version(engine & transmission) changed
    0.8.15 master-ci changed carfw route id : 9a3aa85e86f9c3fc|2022-06-18--15-45-35

log: 
carParams
{'carParams': {'alternativeExperience': 1,
               'carFingerprint': 'mock',
               'carFw': [{'address': 2016,
                          'ecu': 'engine',
                          'fwVersion': b'\xf1\x82OSDAJ6HPXF61\x00\x00\x00\x00',
                          'request': [b'"\xf1\x87\xf1\x89\xf1\x82'],
                          'responseAddress': 2024,
                          'subAddress': 0},
                         {'address': 2017,
                          'ecu': 'transmission',
                          'fwVersion': b'\xf1\x896U3J9051\x00\x00\xf1\x82HOS0G16DS1',
                          'request': [b'"\xf1\x87\xf1\x89\xf1\x82'],
                          'responseAddress': 2025,
                          'subAddress': 0},
                         {'address': 1988,

haram-KONA, 2022-06-20:
Oh, I see. thank you..
However, this phenomenon does not appear in " 0.8.14R3" , and I am curious as to why it suddenly occurred a few days ago in " master-ci ".

Eric recommended this "cmd" to me, and should use this method for the time being, 
or modify "values.py" to use it. 
Is it possible if I request "ADD CARFW" caused by this problem by uploading it to PR?

Eric's cmd : echo 'export SKIP_FW_QUERY=1' >> /data/openpilot/launch_env.sh && echo 'export FINGERPRINT="HYUNDAI KONA HYBRID 2020"' >> /data/openpilot/launch_env.sh

sshane, 2022-06-21:
It looks like the transmission normally responded to `HYUNDAI_VERSION_REQUEST_MULTI`, however after https://github.com/commaai/openpilot/pull/24706 was merged, it started responding to `VOLKSWAGEN_VERSION_REQUEST_MULTI`.

We query VW after Hyundai (first with incorrect offset, second with "correct" Hyundai offset), so I'm thinking what was stopping this from happening before is we were querying the ECUs in common with VW too quickly for the second request with the correct offset, and it either failed to respond in time, gave an error code, or gave a response pending, which we all ignored and kept the Hyundai response. However with correct response pending handling as well as whitelisting VW, it seems like the engine and transmission ECUs on some Hyundais respond to that successfully.

@pd0wm should we revert that VW whitelisting commit and wait for brand filtering (which will fix this and most other brand conflicts), or wait until brand filtering?

sshane, 2022-06-21:
Another idea is we could not throw out the previous returned responses and fingerprint with all for each ECU, then it would reduce the chances of conflicts without the brand filtering. If the FW version response from the VW request doesn't match any Hyundai fingerprints, then we try the other FW version response we got (which is Hyundai and would be successful).

sshane, 2022-06-21:
@dermida looks like another existing issue with missing FW versions on Hyundai.

sshane, 2022-06-21:
@maxtriano let me know if master-ci works for you again

haram-KONA, 2022-06-21:
> 

Hyundai kona hybrid 2020 is good
thank shane ~~~

maxtriano, 2022-06-22:
> @maxtriano let me know if master-ci works for you again

Works now, thanks.

closed 2022-06-22 by maxtriano