commaai/openpilot #24906: Chrysler ACC instant faults with latest master-ci build

ns3284, 2022-06-19:
### Describe the bug

Can't enable ACC with Comma 3 device plugged in. Controls unresponsive on steering wheel until system ultimately faults. Works fine when comma device is unplugged. 

### Which car does this affect?

Chrysler Pacifica 2022

### Provide a route where the issue occurs

4395faadf0a373f2|2022-06-19--15-41-19--0

### openpilot version

0.8.15 Nightly (6/19/2022)

adeebshihadeh, 2022-06-19:
Thanks for reporting this! Looks like it was caused by 0f0b4cac893b41dd81288947158d3aa9cbbfc2fd, so I reverted it on master. @sshane why didn't any tests catch this?

sshane, 2022-06-20:
The segment in test_processes is only available and engaged so flipping them has no effect. We should soon try to get more comprehensive routes, and test more things like doors, blinkers, gas, etc. For now I added a test in test_models

closed 2022-06-22 by sshane