commaai/openpilot #24944: Fix controlsMismatches from 0.8.14-release

adeebshihadeh, 2022-06-22:
```
HONDA ODYSSEY 2018                       0c23034a3a1e7bf4|2022-06-18--19-12-39--17
NISSAN ALTIMA 2020                       78daf6341d75cffb|2022-06-17--09-42-14--2
NISSAN ALTIMA 2020                       78daf6341d75cffb|2022-06-17--09-42-14--3
NISSAN ALTIMA 2020                       78daf6341d75cffb|2022-06-17--10-14-12--3
HONDA CIVIC (BOSCH) 2019                 a18f93cc836f33e4|2022-06-10--16-15-02--39
HONDA CIVIC (BOSCH) 2019                 a18f93cc836f33e4|2022-06-11--06-48-31--4
HONDA CIVIC (BOSCH) 2019                 a18f93cc836f33e4|2022-06-13--09-58-04--0
```

sshane, 2022-06-30:
All Nissans routes seem to be fixed on latest panda master: https://imgur.com/a/NpG14cg

adeebshihadeh, 2022-07-13:
The Nissan appears to be hitting the lag detection in panda somehow. `controls_allowed` is never high for more than two pandaStates frames, and the lag detector runs at 1Hz. We'll need to add some more logging to see what's going on there.

closed 2022-07-14 by adeebshihadeh