commaai/openpilot #24965: C3 restarts randomly due to camera error

khoi, 2022-06-25:
### Describe the bug

My C3 has been restarting randomly even while onroad. I'm opening this issue since the C3 works perfectly fine, no "camera mounted" error. I think maybe this can be a software fix

Context:
- AGNOS 5.1, `master-ci`
- NVME remount was suggested and done, but the issue persists

I managed to change `/var/log` to `/data/log` to persist the log. I got a couple of crashes logged in the file attached

The crash happens on **Jun 25 10:33:50**

- syslog

`Jun 25 10:33:57 tici kernel: [10123.330454] CAM_ERR: CAM-CDM: cam_hw_cdm_submit_gen_irq: 364 cdm test remove bl tag 41 old ctx ffffff825b6201a8 cookie 136706 new ctx ffffff825b629228 cookie 136719`

- kernlog`
`Jun 25 10:33:57 tici kernel: [10123.330454] CAM_ERR: CAM-CDM: cam_hw_cdm_submit_gen_irq: 364 cdm test remove bl tag 41 old ctx ffffff825b6201a8 cookie 136706 new ctx ffffff825b629228 cookie 136719`

[log.zip](https://github.com/commaai/openpilot/files/8984459/log.zip)

Since I don't have comma, here is the `rlog` and `qlog`
[2022-06-25--08-39-59--114.zip](https://github.com/commaai/openpilot/files/8984464/2022-06-25--08-39-59--114.zip)
 

### Provide a route where the issue occurs

2022-06-25--08-39-59--114

### openpilot version

master-ci

khoi, 2022-06-26:
> Jun 25 10:40:01 tici kernel: [   57.122581] CPU: 3 PID: 52540 Comm: dcamera.hevc Tainted: G         C  E   4.9.103+ #3

I think it's the `dcamera`, the crash happens when I open the driver view in settings without going on road

khoi, 2022-06-26:
I re-seated my `dcamera` connectors today, will report back

pd0wm, 2022-06-27:
All of these warnings/error are seen during normal operation, and don't explain a device restart.

You want to look at the `pstore` section of the bootlogs, those contain the last kernel messages before the device rebooted and usually contain the actual crash dump.

But without your device uploading logs to the comma server I can't look into it.

khoi, 2022-06-27:
Sorry for the inconvenience, if u could tell me where to look for the logs, I'll upload them. 

Here are the bootlogs of the sessions
[bootlogs.zip](https://github.com/commaai/openpilot/files/8991291/bootlogs.zip)

pd0wm, 2022-06-27:
None of those contain a pstore entry. You can also find them in `/sys/fs/pstore`.

If you think there is something wrong with your hardware contact support@comma.ai and they can set up a warranty replacement.

khoi, 2022-06-28:
I won't be able to get the `pstore` for the next 3 weeks as I'm out of town. I'll closing this for now as this is likely to be hardware issue. Will reopen if needed. Thanks

khoi, 2022-07-24:
Reopening. 

> All of these warnings/error are seen during normal operation, and don't explain a device restart.

I don't think this is the case, I have another healthy C3 running the same code, and it doesn't have those lines. Also those lines repeat everytime the device reboots (attached syslog where it happens multiple times during 10 mins)

[syslog.zip](https://github.com/commaai/openpilot/files/9175341/syslog.zip)

Excerpts from the log

```
Jul 22 11:43:12 tici kernel: [   57.399915] i2c_geni 890000.i2c: i2c error :-107
...
Jul 22 11:43:13 tici kernel: [   57.801122] CAM_INFO: CAM-CRM: cam_req_mgr_open: 130 cam_req_mgr_open:130 disalbe lpm
Jul 22 11:43:13 tici kernel: [   57.801142] CAM_INFO: CAM-SENSOR: cam_pm_qos_update_request: 117 cam_pm_qos_update_request: update request 100
Jul 22 11:43:13 tici kernel: [   57.918936] CAM_INFO: CAM-SENSOR: cam_sensor_driver_cmd: 841 CAM_ACQUIRE_DEV Success, sensor_id:0x354,sensor_slave_addr:0x20
Jul 22 11:43:13 tici kernel: [   57.988361] CAM_INFO: CAM-SENSOR: cam_sensor_driver_cmd: 768 Probe success,slot:1,slave_addr:0x30,sensor_id:0x354
Jul 22 11:43:13 tici kernel: [   57.994812] CAM_INFO: CAM: cam_res_mgr_shared_pinctrl_put: 273 Need hold put this pinctrl
Jul 22 11:43:13 tici kernel: [   58.048188] CAM_INFO: CAM-SENSOR: cam_sensor_driver_cmd: 841 CAM_ACQUIRE_DEV Success, sensor_id:0x354,sensor_slave_addr:0x30
Jul 22 11:43:13 tici kernel: [   58.118340] CAM_INFO: CAM-SENSOR: cam_sensor_driver_cmd: 768 Probe success,slot:0,slave_addr:0x20,sensor_id:0x354
Jul 22 11:43:13 tici kernel: [   58.124793] CAM_INFO: CAM: cam_res_mgr_shared_pinctrl_put: 273 Need hold put this pinctrl
Jul 22 11:43:14 tici kernel: [   58.209991] msm_vidc: info: Opening video instance: 0000000000000000, 0
Jul 22 11:43:14 tici kernel: [   58.214760] msm_vidc: info: Opening video instance: 0000000000000000, 0
Jul 22 11:43:14 tici kernel: [   58.218366] msm_vidc: info: Opening video instance: 0000000000000000, 0
Jul 22 11:43:14 tici kernel: [   58.248530] msm_vidc: info: Opening video instance: 0000000000000000, 0
...
Jul 22 11:43:14 tici kernel: [   58.479416] PC is at vb2_warn_zero_bytesused.isra.10.part.11+0x24/0x68
Jul 22 11:43:14 tici kernel: [   58.479425] LR is at __fill_vb2_buffer+0x4d0/0x508
Jul 22 11:43:14 tici kernel: [   58.479434] pc : [<ffffff891ca990bc>] lr : [<ffffff891ca995d0>] pstate: 20400145
Jul 22 11:43:14 tici kernel: [   58.479442] sp : ffffffd6cec2f7e0
Jul 22 11:43:14 tici kernel: [   58.479450] x29: ffffffd6cec2f7e0 x28: ffffff891efcbbd8 
Jul 22 11:43:14 tici kernel: [   58.479472] x27: 0000000000000030 x26: ffffffd70dc15000 
Jul 22 11:43:14 tici kernel: [   58.479494] x25: ffffffd6cec2f8e8 x24: ffffffd70dc15000 
Jul 22 11:43:14 tici kernel: [   58.479516] x23: 0000000000000000 x22: ffffffd6cec2f8e8 
Jul 22 11:43:14 tici kernel: [   58.479538] x21: 0000000000000000 x20: ffffffd72f1f8000 
Jul 22 11:43:14 tici kernel: [   58.479560] x19: ffffffd72f1f8000 x18: 0000000000000000 
Jul 22 11:43:14 tici kernel: [   58.479582] x17: 0000007f9a666930 x16: ffffff891c2552d0 
Jul 22 11:43:14 tici kernel: [   58.479603] x15: 000056790b54c7a8 x14: 0000000000000000 
Jul 22 11:43:14 tici kernel: [   58.479625] x13: 0000000000000000 x12: 0000000000000000 
Jul 22 11:43:14 tici kernel: [   58.479647] x11: 0000000000000000 x10: 0000000000000000 
Jul 22 11:43:14 tici kernel: [   58.479668] x9 : 0000000000000000 x8 : ffffffd6cec2f918 
Jul 22 11:43:14 tici kernel: [   58.479690] x7 : 0000000000000000 x6 : 000000000000003f 
Jul 22 11:43:14 tici kernel: [   58.479712] x5 : 0000000000000000 x4 : 0000000000000030 
Jul 22 11:43:14 tici kernel: [   58.479733] x3 : 0000000000000000 x2 : 0000000000000008 
Jul 22 11:43:14 tici kernel: [   58.479755] x1 : 0000000000000001 x0 : ffffff891efcb000 
Jul 22 11:43:14 tici kernel: [   58.479778] 
Jul 22 11:43:14 tici kernel: [   58.479778] PC: 0xffffff891ca9907c:
Jul 22 11:43:14 tici kernel: [   58.479785] 907c  a8c27bfd d65f03c0 2a1403e1 97ffffce a94153f3 a8c27bfd d65f03c0 a9be7bfd
Jul 22 11:43:14 tici kernel: [   58.479857] 909c  910003fd f9000bf3 aa0003f3 aa1e03e0 d503201f d0012980 52800021 392f7001
Jul 22 11:43:14 tici kernel: [   58.479927] 90bc  d4210000 b0006e80 91000000 97dcee68 f9400260 39406000 361000e0 b0006e80
Jul 22 11:43:14 tici kernel: [   58.479997] 90dc  91014000 97dcee62 f9400bf3 a8c27bfd d65f03c0 b0006e80 91022000 97dcee5c
Jul 22 11:43:14 tici kernel: [   58.480068] 
Jul 22 11:43:14 tici kernel: [   58.480068] LR: 0xffffff891ca99590:
Jul 22 11:43:14 tici kernel: [   58.480076] 9590  b9403e60 17ffff3a b9400a60 35ffe480 b9401b20 b9001720 17ffff22 12802701
Jul 22 11:43:14 tici kernel: [   58.480146] 95b0  52800015 0a010000 b9022280 17ffff7a b900173f 17ffff59 aa1403e0 97fffeb3
Jul 22 11:43:14 tici kernel: [   58.480216] 95d0  f9400280 39406000 3617f100 b8776b00 17ffff49 aa1403e0 97fffeac f9400280
Jul 22 11:43:14 tici kernel: [   58.480286] 95f0  39406000 3617fd20 b9400a60 b9001720 17ffff0c 00000000 a9be7bfd 910003fd
Jul 22 11:43:14 tici kernel: [   58.480357] 
Jul 22 11:43:14 tici kernel: [   58.480357] SP: 0xffffffd6cec2f7a0:
Jul 22 11:43:14 tici kernel: [   58.480364] f7a0  1ca995d0 ffffff89 cec2f7e0 ffffffd6 1ca990bc ffffff89 20400145 00000000
Jul 22 11:43:14 tici kernel: [   58.480434] f7c0  2c83fe00 ffffffd7 00000096 00000000 ffffffff 0000007f 18eff120 ffffff80
Jul 22 11:43:14 tici kernel: [   58.480504] f7e0  cec2f800 ffffffd6 1ca995d0 ffffff89 cec2fcf8 ffffffd6 1c1b29fc ffffff89
[... cut ...]
Jul 22 11:43:14 tici kernel: [   58.480892] [<ffffff891ca7626c>] v4l2_ioctl+0xbc/0x108
Jul 22 11:43:14 tici kernel: [   58.480904] [<ffffff891c254b84>] do_vfs_ioctl+0xc4/0x810
Jul 22 11:43:14 tici kernel: [   58.480913] [<ffffff891c255360>] SyS_ioctl+0x90/0xa8
Jul 22 11:43:14 tici kernel: [   58.480926] [<ffffff891c083f80>] el0_svc_naked+0x34/0x38
Jul 22 11:43:14 tici kernel: [   58.480936] use of bytesused == 0 is deprecated and will be removed in the future,
Jul 22 11:43:14 tici kernel: [   58.480948] use the actual size instead.
```

khoi, 2022-07-24:
my device is banned (`ec07911a91f863f3`) for known reason, I checked `pstore` there is nothing there. But if u want to the device to uploads stuff again, pls unban it, at least for a bit so I have the chance to upload those files.

khoi, 2022-07-31:
I took C3 to my office and use 

`STARTED=1 ./scripts/launch_corolla.sh`

Same behavior, it reboots after awhile.  

[image]

[image]

khoi, 2022-07-31:
Some interesting kern.log lines:

```
Jul 31 14:54:34 tici kernel: [ 1855.332660] i2c_geni 890000.i2c: i2c error :-107
Jul 31 14:54:35 tici kernel: [ 1856.556215] msm_vidc:  err: got buffer back with error ea5e8
Jul 31 14:54:42 tici kernel: [ 1863.716893] CAM_ERR: CAM-CDM: cam_hw_cdm_submit_gen_irq: 364 cdm test remove bl tag 48 old ctx ffffff953da201a8 cookie 140 new ctx ffffff953da29228 cookie 153
```

adeebshihadeh, 2023-07-04:
Don't see any recent routes. Still having this issue?

khoi, 2023-07-17:
@adeebshihadeh I bought a new C3 since that one is broken

khoi, 2023-07-17:
I guess we can close it

closed 2023-07-17 by khoi