commaai/openpilot #25004: Toyota: Potential EPS belt wear out

Smartype, 2022-06-30:
### Describe the bug

This is a Toyota Avalon 2018 non-hybrid car. With stock LDA and low speed limit ACC.   The mileage is about 40000km. installed  openpilot for about one year.  When take apart the EPS motor from the steering rack, found that the belt wore out. The motor gear  wore out too. 
[image]

[image]

### Which car does this affect?

Toyota Avalon 2019 Chinese

### Provide a route where the issue occurs

Not available

### openpilot version

0.8.7

Smartype, 2022-06-30:
The car own bought a new Avalon 291 steering rack (with tss 2.x support, lta). And he said the mechanical struct/belt is the same.

zorrobyte, 2022-06-30:
@Smartype It makes sense, the car was not designed for constant steering. I'd consider this a wear item if using OP on this model moving forward. Although it would be nice for comma to add a note to the vehicle list, I doubt they would.

Good find!

Smartype, 2022-06-30:
> @Smartype It makes sense, the car was not designed for constant steering. I'd consider this a wear item if using OP on this model moving forward. Although it would be nice for comma to add a note to the vehicle list, I doubt they would.
> 
> 
> 
> Good find! Makes me wonder if this would have happened to the Rav if I would have kept it.

The new Avalon 2021 has LTA support. (I guess it is supposed to do steering whenever possible, am I right?) 

And it has the same steering rack.(maybe with a different FW)

jyoung8607, 2022-06-30:
> @Smartype It makes sense, the car was not designed for constant steering. I'd consider this a wear item if using OP on this model moving forward. Although it would be nice for comma to add a note to the vehicle list, I doubt they would.

Important parts of those pictures are blurry, but I don't see how a neoprene-fiberglass or urethane-polyester belt wipes out a metal gear like that, especially without showing severe wear, which it doesn't. We see the belt has lost some chunks, but the parts we can see of the belt are basically there. What's been lost are a few relatively meaty chunks from some sort of sudden trauma, not a gradual fine deposit of powder-like belt material like you'd see from wear over time. Maybe some metallic FOD got loose in there and went dancing around?

geohot, 2022-06-30:
@zorrobyte If I see more stuff like "Although it would be nice for comma to add a note to the vehicle list, I doubt they would.", you'll be banned from our github. If there's a real issue pertaining to a model of car, we'd be happy to note that after we have real evidence. Assuming bad faith = ban.

Now, I can't see why this would affect specifically the Avalon, or have anything to do with openpilot at all. The EPS motor is constantly applying torque to the wheel when the human is steering also. I would guess that something else happened to the Avalon, the main issue with a sample size of 1.

pd0wm, 2022-06-30:
@Smartype how did you notice this issue? Did this user loose power steering, or get some form of warning/DTC?

Smartype, 2022-07-01:
> @Smartype how did you notice this issue? Did this user loose power steering, or get some form of warning/DTC?

The car owner was trying to upgrade to tss 2.5. So he has lots of parts to be replaced. EPS motor is one of then. He accidentally found this issue.

adeebshihadeh, 2022-08-26:
I'm going to close this. There's nothing substantive for us to look into here beyond "here's a blurry picture of a failed EPS driven by openpilot, and look at this perfect one without openpilot." Feel free to open a new one if anything ends up pointing to openpilot.

closed 2022-08-26 by adeebshihadeh