commaai/openpilot #25012: Toyota RAV4 hybrid 2020, unsupported car Issue

CPontes, 2022-07-01:
### Describe the bug

When connect comma.ai, to my Toyota RAV4 hybrid 2020 it brings the message that it is an unsupported car. 
I changed the cables but the error persists. 
I would be glad if someone can help on this issue.

### Which car does this affect?

Toyota RAV4 hybrid 2020

### Provide a route where the issue occurs

cb7a33a4e4f4116b|2022-06-30--23-11-12--0

### openpilot version

0.8.13.1

sshane, 2022-07-02:
It looks like the device isn't able to query any ECUs for fingerprinting, is your comma power plugged in (cable running to OBD port)?

CPontes, 2022-07-02:
> It looks like the device isn't able to query any ECUs for fingerprinting, is your comma power plugged in (cable running to OBD port)?

Thank you so much for the help. Yes, it’s all connected, it was working fine on my Corolla and when I connected on the rav4 it didn’t work. Is there something you suggest to do to try do solve this?

luisgrases, 2022-07-03:
My RAV4 2019 was not recognized so I had to go through this process: https://github.com/commaai/openpilot/wiki/Fingerprinting

pd0wm, 2022-07-04:
> > It looks like the device isn't able to query any ECUs for fingerprinting, is your comma power plugged in (cable running to OBD port)?
> 
> Thank you so much for the help. Yes, it’s all connected, it was working fine on my Corolla and when I connected on the rav4 it didn’t work. Is there something you suggest to do to try do solve this?
> 
> I think I sent the wrong one,
> 
> Can you check this one?
> 
> cb7a33a4e4f4116b|2022-07-02--11-35-02--0
> 
> [image]

Looks like a missing engine FW:
```
cb7a33a4e4f4116b|2022-07-02--11-35-02
Old style: TOYOTA RAV4 HYBRID 2019 Vin JTMDW3FV2LD527443
New style (exact): set()
New style (fuzzy): set()
  (Ecu.esp, 0x7b0, None): [b'F152642521\x00\x00\x00\x00\x00\x00'],
  (Ecu.eps, 0x7a1, None): [b'8965B42171\x00\x00\x00\x00\x00\x00'],
  (Ecu.engine, 0x700, None): [b'\x018966342W7000\x00\x00\x00\x00'],
  (Ecu.fwdRadar, 0x750, 0xf): [b'\x018821F3301400\x00\x00\x00\x00'],
  (Ecu.fwdCamera, 0x750, 0x6d): [b'\x028646F4203500\x00\x00\x00\x008646G2601200\x00\x00\x00\x00'],
Mismatches
(0x700, None) - b'\x018966342W7000\x00\x00\x00\x00'
```

However, you're still on 0.8.13 due to the comma two no longer receiving updates. So there might be more missing fingerprints for you.

sshane, 2022-07-07:
Added your missing fingerprint to upstream openpilot, you can try to cherry-pick it onto your local branch, or hard-code the fingerprint. Feel free to ask on the Discord to perform either of those things. PR: https://github.com/commaai/openpilot/pull/25057

closed 2022-07-07 by sshane