commaai/openpilot #25028: hyundai: fix carState.stockAeb/carState.stockFcw

gregjhogan, 2022-07-04:
The current logic isn't correct
https://github.com/commaai/openpilot/blob/0c95493dc0cbc5d3dec59a50ac5476573041942d/selfdrive/car/hyundai/carstate.py#L112-L117

stock seems to use `CF_VSM_Warn == 1` for warning the driver and higher values for braking, although it is possible to command braking even when `CF_VSM_Warn == 1`

stock definitely uses `CF_VSM_Warn == 2` with `CF_VSM_DecCmdAct = 1` as well as `CF_VSM_Warn == 3` with `FCA_CmdAct = 1` to command braking (and in both cases uses `CR_VSM_DecCmd` for the amount of braking requested)

Here is my best guess but there still may be edge cases where it doesn't brake until you start pressing the brake pedal or something
* stockFcw = `CF_VSM_Warn != 0 and CF_VSM_DecCmdAct == 0 and FCA_CmdAct == 0`
* stockAeb = `CF_VSM_Warn != 0 and (CF_VSM_DecCmdAct == 0 or FCA_CmdAct == 0)`

Example routes:
02c45f73a2e5c6e9|2020-06-24--12-20-18
c7d0d7dde77d49ce|2022-06-24--19-02-59

sunnyhaibin, 2022-07-04:
This is what I have noticed on the 2021 Hyundai Sonata as well.
[image]

sshane, 2022-07-13:
`CF_VSM_DecCmdAct != 0 or FCA_CmdAct != 0` you mean?

sshane, 2022-08-08:
This user also got a FCW: 6111bd7df5043394|2022-08-08--09-48-29--15

[image]

PR #25152 (2022-10-08): hyundai: fix logging stock AEB events
fix #25028
closed 2022-10-08 by sshane