commaai/openpilot #25096: EV6: VIN returns on non-functional address

sshane, 2022-07-09:
Responses from all addresses from 0x700 + 256:

```python
(1971, None) uds b'1.08.00'
(2002, None) uds b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\:
(1990, None) uds b'KNDC44LA2N5030320'
(2022, None) uds b'\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\;
(2020, None) uds b'CVEV-- BMS
(2018, None) uds b'KNDC44LA2N5030320'
```

sshane, 2022-07-09:
Interestingly, I see these addresses in this document: https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/master/vehicle/OVMS.V3/components/vehicle_kianiroev/src/vehicle_kianiroev.cpp

1990 is `Cluster. ODO`
2018 is `VMCU - VIN`

@gregjhogan think we need another brand-specific request for this car, or does it really respond on a non-standard address?

sshane, 2022-07-09:
I checked a Niro EV actually, and it has the same problem (it responded with `'DEEV__ BMS \x00\x00\x00\x00\x00\x00'`), maybe this is a problem with all the HKG EVs.

gregjhogan, 2022-07-09:
Mixing sending requests for functional and physical addresses kind of sounds like a mess (you can't do it in parallel).  It seems possible that all supported vehicles that are not electric currently respond with the VIN on 0x7e8 meaning we could potentially switch it to only querying two addresses: 0x7e0 (engine) and 0x7e2 (vehicle motor control unit) and query them in parallel.

sshane, 2022-07-12:
This seems to work! EV6 responses to both UDS and OBD queries on 0x7e2, Optima and RAV4 on 0x7e0

sshane, 2022-07-12:
Looks like the Civic only responds on the 29 bit functional address. Can we not query that in parallel?

gregjhogan, 2022-07-12:
I believe the only thing we cannot do in parallel is:
* sub addresses (I believe we have found this doesn't work for Toyota)
* physical and functional addresses (functional address response handling has to assume all response messages pertain to the functional address request)

sshane, 2022-07-12:
Aren't functional addresses labeled by query type? Could we read the response data to know which tx addr is for an rx addr? We could use that, but that's extra complexity

We could still make it work as is for VIN, as we only have one functional 29 bit address. Might take some changes to IsoTpParallelQuery though

gregjhogan, 2022-07-12:
With some work you could probably do 29 bit functional and 11 bit physical addresses in parallel since they don't overlap.
It is harder to say if you can handle a physical address that also responds to a functional address (the response address is the same for both and the ECU could do something dumb like respond to both putting duplicate messages on the CAN bus that will break expected sequence)

closed 2022-08-30 by sshane