commaai/openpilot #25112: scons failing on mac bootstrap

herman5, 2022-07-11:
### Describe the bug

On macOS 12.4, scons fails to build after a fresh openpilot install and running `$ tools/mac_setup.sh`

```
$ cd openpilot && pipenv shell
$ scons -u -j$(nproc)

Warning: Please note that the following versions of CasADi  are officially supported: 3.5.5 or 3.5.4 or 3.5.3 or 3.5.2 or 3.5.1 or 3.4.5 or 3.4.0.
 If there is an incompatibility with the CasADi generated code, please consider changing your CasADi version.
Version 3.5.4+ currently in use.
sh: /Users/dylanherman/openpilot/third_party/acados/Darwin/t_renderer: Bad CPU type in executable

Exception: Rendering of main.in.c failed!

Attempted to execute OS command:
/Users/dylanherman/openpilot/third_party/acados/Darwin/t_renderer '/Users/dylanherman/openpilot/pyextra/acados_template/c_templates_tera/*' 'main.in.c' '/Users/dylanherman/openpilot/selfdrive/controls/lib/longitudinal_mpc_lib/acados_ocp_long.json' 'main_long.c'

Exiting.

scons: *** [selfdrive/controls/lib/longitudinal_mpc_lib/c_generated_code/Makefile] Error 1
scons: building terminated because of errors.
```

Opening as an issue as it's been referenced in [past discussions](https://github.com/commaai/openpilot/discussions/2524?sort=top#discussioncomment-2952715) as well as in [#dev-openpilot](https://discord.com/channels/469524606043160576/524594418628558878/980495827120304259) with no resolution.

This was ran on a macbook with Intel CPUs.

### OS Version

macOS 12.4

### openpilot version or commit

eb17291ca13035b40d653da107653d4420517aaa

LoganxDev, 2022-07-23:
Yep, I have a branch I've started working on supporting intel macs with. They only have the arm executable for macOS at the moment.

herman5, 2022-07-24:
@LoganxDev thanks! If you want to push that branch, maybe we can track progress here.

LoganxDev, 2022-07-27:
[wip branch](https://github.com/LoganxDev/openpilot/tree/llas/fix/scons/add-macosx86-support)

There's the branch I'm using, the current issue is some ffmpeg lib headers aren't being found even though the libs are in the lib path in the root Sconstruct.

incognitojam, 2023-01-21:
There have been a few PRs to fix builds on macOS, do you want to check if this is still an issue on the latest master?

haraschax, 2023-05-23:
Mac is being worked on

closed 2023-05-23 by haraschax