commaai/openpilot #25160: Ensure CANParser counter/checksums checks are the same as panda

sshane, 2022-07-13:
For example, `SCM_BUTTONS`'s counter started returning garbage values (or issue with a cable) and only panda noticed it, causing a controls mismatch.

We need to change how the checks work in the CANParser, a counter and checksum issue uses the same thresholds as the TIMEOUT error, and we don't differentiate them.

http://data-raw.internal/c390b60ad96d2ff2/2022-07-01--07-40-21/4/rlog.bz2

adeebshihadeh, 2022-07-15:
This case really looks like a bad cable. We also need a better check than a threshold on bad counters, as some mildly faulty cables can slip through like in this case.

sshane, 2022-08-20:
- [ ] Do same check for checksum now

adeebshihadeh, 2022-11-03:
@sshane is this done?

sshane, 2022-11-03:
Made the counter checks match, I'll check the checksum checks. Might already match

adeebshihadeh, 2024-01-05:
@sshane this should be done with the new fuzzing tests right?

closed 2024-01-09 by adeebshihadeh

sshane, 2024-01-09:
I believe so.