commaai/openpilot #25165: build failure in dbc.cc

bulvara, 2022-07-13:
### Describe the bug

```
opendbc/can/dbc.cc:200:50: error: no template named 'ostream_iterator' in namespace 'std'; did you mean 'ostreambuf_iterator'?
      std::copy(words.begin(), words.end(), std::ostream_iterator<std::string>(s, " "));
                                            ~~~~~^~~~~~~~~~~~~~~~
                                                 ostreambuf_iterator
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/streambuf_iterator.h:248:11: note: 'ostreambuf_iterator' declared here
    class ostreambuf_iterator
          ^
opendbc/can/dbc.cc:200:45: error: no matching constructor for initialization of 'std::ostreambuf_iterator<std::string>' (aka 'ostreambuf_iterator<basic_string<char>>')
      std::copy(words.begin(), words.end(), std::ostream_iterator<std::string>(s, " "));
                                            ^                                  ~~~~~~
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/streambuf_iterator.h:283:7: note: candidate constructor not viable: requires single argument '__s', but 2 arguments were provided
      ostreambuf_iterator(ostream_type& __s) _GLIBCXX_USE_NOEXCEPT
      ^
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/streambuf_iterator.h:287:7: note: candidate constructor not viable: requires single argument '__s', but 2 arguments were provided
      ostreambuf_iterator(streambuf_type* __s) _GLIBCXX_USE_NOEXCEPT
      ^
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/streambuf_iterator.h:248:11: note: candidate constructor (the implicit copy constructor) not viable: requires 1 argument, but 2 were provided
    class ostreambuf_iterator
          ^
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/streambuf_iterator.h:248:11: note: candidate constructor (the implicit move constructor) not viable: requires 1 argument, but 2 were provided
In file included from opendbc/can/dbc.cc:2:
In file included from /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/filesystem:44:
In file included from /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/fs_fwd.h:35:
In file included from /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/system_error:41:
In file included from /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/stdexcept:39:
In file included from /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/string:40:
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/char_traits.h:152:16: error: no matching conversion for static_cast from 'const __gnu_cxx::char_traits<std::basic_string<char>>::char_type' (aka 'const std::basic_string<char>') to '__gnu_cxx::char_traits<std::basic_string<char>>::int_type' (aka 'unsigned long')
      { return static_cast<int_type>(__c); }
               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/streambuf:436:27: note: in instantiation of member function '__gnu_cxx::char_traits<std::basic_string<char>>::to_int_type' requested here
            __ret = traits_type::to_int_type(__c);
                                 ^
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/streambuf_iterator.h:295:36: note: in instantiation of member function 'std::basic_streambuf<std::basic_string<char>>::sputc' requested here
            _Traits::eq_int_type(_M_sbuf->sputc(__c), _Traits::eof()))
                                          ^
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/stl_algobase.h:385:18: note: in instantiation of member function 'std::ostreambuf_iterator<std::basic_string<char>>::operator=' requested here
              *__result = *__first;
                        ^
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/stl_algobase.h:495:22: note: in instantiation of function template specialization 'std::__copy_move<false, false, std::random_access_iterator_tag>::__copy_m<std::basic_string<char> *, std::ostreambuf_iterator<std::basic_string<char>>>' requested here
                              _Category>::__copy_m(__first, __last, __result);
                                          ^
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/stl_algobase.h:522:19: note: in instantiation of function template specialization 'std::__copy_move_a2<false, std::basic_string<char> *, std::ostreambuf_iterator<std::basic_string<char>>>' requested here
    { return std::__copy_move_a2<_IsMove>(__first, __last, __result); }
                  ^
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/stl_algobase.h:530:8: note: in instantiation of function template specialization 'std::__copy_move_a1<false, std::basic_string<char> *, std::ostreambuf_iterator<std::basic_string<char>>>' requested here
                std::__copy_move_a1<_IsMove>(std::__niter_base(__first),
                     ^
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/stl_algobase.h:619:19: note: in instantiation of function template specialization 'std::__copy_move_a<false, __gnu_cxx::__normal_iterator<std::basic_string<char> *, std::vector<std::basic_string<char>>>, std::ostreambuf_iterator<std::basic_string<char>>>' requested here
      return std::__copy_move_a<__is_move_iterator<_II>::__value>
                  ^
opendbc/can/dbc.cc:200:12: note: in instantiation of function template specialization 'std::copy<__gnu_cxx::__normal_iterator<std::basic_string<char> *, std::vector<std::basic_string<char>>>, std::ostreambuf_iterator<std::basic_string<char>>>' requested here
      std::copy(words.begin(), words.end(), std::ostream_iterator<std::string>(s, " "));
           ^
/usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/12.1.0/../../../../include/c++/12.1.0/bits/basic_string.h:936:7: note: candidate function
      operator __sv_type() const noexcept
      ^
3 errors generated.
scons: *** [opendbc/can/dbc.os] Error 1
```

### Provide a route where the issue occurs

scons .

### openpilot version

git master

pd0wm, 2022-07-14:
Can you provide more information about your environment?

bulvara, 2022-07-14:
ArchLinux with gcc 12.1

Because `git submodule update --init --recursive` doesn't work I'm
- * cloning 'git+https://github.com/commaai/opendbc.git'
- * rm -rf opendbc
- * ln -s ../opendbc ./opendbc

pd0wm, 2022-07-14:
`dbc.cc` is probably missing an `#include <iterator>`. We don't officially support gcc, but if that fixes it for you feel free to make a PR.

closed 2022-07-14 by pd0wm