commaai/openpilot #25173: Cross-compile with arm-none-eabi-gcc for the panda fails (include_next)

bulvara, 2022-07-14:
### Describe the bug

I'm trying to package OpenPilot for ArchLinux. After installing `arm-none-eabi-gcc` for the cross-compilation it fails:
```
arm-none-eabi-gcc -o panda/board/bootstub-panda.o -c -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -T./openpilot/panda/board/stm32fx/stm32fx_flash.ld -std=gnu11 -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA -DALLOW_DEBUG -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc -Ipanda -Ipanda/board panda/board/bootstub.c
In file included from panda/board/stm32fx/inc/core_cm4.h:34,
                 from panda/board/stm32fx/inc/stm32f413xx.h:197,
                 from panda/board/stm32fx/inc/stm32f4xx.h:189,
                 from panda/board/stm32fx/stm32fx_config.h:2,
                 from panda/board/config.h:34,
                 from panda/board/bootstub.c:7:
/usr/lib/gcc/arm-none-eabi/12.1.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
      |                ^~~~~~~~~~
compilation terminated.
```
/usr/lib/gcc/{arm-none-eabi – _x86_64-pc-linux-gnu_ version is identical}/12.1.0/include/stdint.h looks like this:
```
#ifndef _GCC_WRAP_STDINT_H
#if __STDC_HOSTED__
# if defined __cplusplus && __cplusplus >= 201103L
#  undef __STDC_LIMIT_MACROS
#  define __STDC_LIMIT_MACROS
#  undef __STDC_CONSTANT_MACROS
#  define __STDC_CONSTANT_MACROS
# endif
# include_next <stdint.h>
#else
# include "stdint-gcc.h"
#endif
#define _GCC_WRAP_STDINT_H
#endif
```

### Provide a route where the issue occurs

scons .

### openpilot version

master (>0.8.14)

pd0wm, 2022-07-14:
Github issues are not the place for compile errors on non supported OSes. Please join discord.comma.ai and you can ask you questions in the #dev-openpilot channel.

bulvara, 2022-07-14:
@[pd0wm](https://github.com/pd0wm) I can't a verification for mail address from Discord.

My issue seems to be more universal: https://github.com/commaai/openpilot/issues/24500