commaai/openpilot #25174: Legacy python "future_fstrings" in acados_template

bulvara, 2022-07-14:
### Describe the bug

```
cd ./openpilot/selfdrive/controls/lib/lateral_mpc_lib && python3 lat_mpc.py
Traceback (most recent call last):
  File "./openpilot/selfdrive/controls/lib/lateral_mpc_lib/lat_mpc.py", line 12, in <module>
    from pyextra.acados_template import AcadosModel, AcadosOcp, AcadosOcpSolver
  File "./openpilot/pyextra/acados_template/__init__.py", line 35, in <module>
    from .generate_c_code_explicit_ode import *
  File "./openpilot/pyextra/acados_template/generate_c_code_explicit_ode.py", line 36, in <module>
    from .utils import ALLOWED_CASADI_VERSIONS, is_empty, casadi_version_warning
  File "./openpilot/pyextra/acados_template/utils.py", line 0
SyntaxError: unknown encoding: future_fstrings
scons: *** [selfdrive/controls/lib/lateral_mpc_lib/c_generated_code/Makefile] Error 1
scons: building terminated because of errors.
```

### Provide a route where the issue occurs

scons .

### openpilot version

master (>0.8.14)

bulvara, 2022-07-15:
@adeebshihadeh Why have you closed this?

adeebshihadeh, 2022-07-15:
This isn't a bug. Our issue tracker isn't for help getting openpilot buliding.