commaai/openpilot #25182: Ioniq 5 - CAN Bus Disconnected: Likely Faulty Cable

G80Tampa, 2022-07-15:
### Describe the bug

Error, CAN Bus Disconnected: Likely Faulty Cable

[image]
[image]

### Which car does this affect?

Hyundai Ioniq 5 2022

### Provide a route where the issue occurs

51b07e62f9b64fee|2022-07-14--19-25-33--1

### openpilot version

0.8.15

spectre3ooo, 2022-07-19:
Here's another drive from my car:
45dfe516db6622f8|2022-07-19--00-10-13
But, i'm getting a camera malfunction error on my C3 now.

hoomoose, 2022-07-20:
I get this exact same error with my Kia EV6.
Route: 68d6a96e703c00c9%7C2022-07-19--16-53-15

sunnyhaibin, 2022-07-28:
The Ioniq 5 from the issue's OP confirmed that this specific Ioniq 5 does not have the `LKAS` message `0x2a4`.

spectre3ooo, 2022-07-28:
> The Ioniq 5 from the issue's OP confirmed that this specific Ioniq 5 does not have the `LKAS` message `0x2a4`.

In this case, would you tell it not to look for 0x2a4, and instead use the LFA messages?

closed 2022-08-12 by adeebshihadeh