commaai/openpilot #25194: Early and harsh braking on 0.8.15 ( master-ci)

adnanlanewla, 2022-07-17:
### Describe the bug

I have had many instances where OP brakes quite early and quite harsh. I mainly notice this in stop n go traffic on Highway.

Here's the route where this is happening

3f3a48346848df21|2022-07-16--18-12-10--21

3f3a48346848df21|2022-07-16--18-12-10--23

3f3a48346848df21|2022-07-16--18-12-10--29

### Provide a route where the issue occurs

3f3a48346848df21|2022-07-16--18-12-10--21

### openpilot version

0.8.15

sshane, 2022-07-19:
It looks like your car was following the plan pretty closely, the planner just decided to brake pretty late, like due to aLeadTau:

[image]

sshane, 2022-07-19:
Stock ACC wanted to brake 1.5 seconds earlier here, and with much less total deceleration:
[image]

adnanlanewla, 2022-07-19:
> It looks like your car was following the plan pretty closely, the planner just decided to brake pretty late, likely due to aLeadTau. The lead decelerated at a max of -2.7 m/s/s however openpilot requested -3.5 m/s/s because it was a bit late to start braking (more than a second after lead):
> 
> 29th segment: [image]
> 
> @HaraldSchafer Looks like aLeadTau was added in one around here, and the kalman filter doesn't look too sophisticated compared to now, has it improved at all since it was introduced? I drove on no aLeadTau on my road trip and it was fine, braked and accelerated earlier and smoother:
> 
> https://github.com/commaai/one/blob/1032938cc6ac5628d66b3262bf4e998c25338923/selfdrive/controls/lib/radar_helpers.py

Hi Shane, 
That's the thing I have noticed this in almost all of my drives either OP brakes quite early and quite hard or it brakes a little late and still quite hard

HaraldSchafer, 2022-08-06:
Linking to https://github.com/commaai/openpilot/issues/25295

closed 2022-08-06 by HaraldSchafer