commaai/openpilot #25195: Car no longer FP correctly

khoi, 2022-07-18:
### Describe the bug

My Hyundai Santa Fe 2022 is no longer FP correctly, after checking `carParams`, the value seems to be changed? Do I need to update them?

https://github.com/commaai/openpilot/pull/23718/files

### Provide a route where the issue occurs

ec07911a91f863f3|2022-07-18--11-18-16

### openpilot version

master-ci

sshane, 2022-07-18:
Besides the recent PR to use the brand FW response, looks like your car is also returning a new transmission FW not in the database. It would have at least fuzzy fingerprinted with that added

sshane, 2022-07-18:
Want to try this branch? https://github.com/commaai/openpilot/pull/25205

sshane, 2022-07-19:
Fixed in https://github.com/commaai/openpilot/pull/25205 and https://github.com/commaai/openpilot/pull/25204

closed 2022-07-19 by sshane