commaai/openpilot #25201: No LTE on C3 own SIM

Piotrreeqq, 2022-07-18:
### Describe the bug

I spotted a problem with own sim card on agnos 5.2, previously I was using version 4 and everything was working fine, now there is a problem with connection. 

```json
Jul 18 17:38:13 tici NetworkManager[4009]: <info>  [1658158693.0869] device (cdc-wdm0): state change: prepare -> failed (reason 'unknown', sys-iface-state: 'managed')
Jul 18 17:38:13 tici NetworkManager[4009]: <warn>  [1658158693.0904] device (cdc-wdm0): Activation: failed for connection 'lte'
Jul 18 17:38:13 tici NetworkManager[4009]: <info>  [1658158693.0914] device (cdc-wdm0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Jul 18 17:38:13 tici NetworkManager[4009]: <info>  [1658158693.0944] device (cdc-wdm0): Activation: starting connection 'lte' (6586fb1f-4a66-46d4-9164-0100b73750ee)
Jul 18 17:38:13 tici NetworkManager[4009]: <info>  [1658158693.0948] device (cdc-wdm0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Jul 18 17:38:14 tici NetworkManager[4009]: <warn>  [1658158694.0107] modem-broadband[cdc-wdm0]: failed to connect modem: Multiple PDN connection to same APN not allowed
Jul 18 17:38:14 tici NetworkManager[4009]: <info>  [1658158694.0107] device (cdc-wdm0): state change: prepare -> failed (reason 'unknown', sys-iface-state: 'managed')

### Provide a route where the issue occurs

84f90dbb726265da

### openpilot version

0.8.16

adeebshihadeh, 2022-07-18:
Can you provide some more information, like carrier, ModemManager logs, NetworkManager logs, etc?

Piotrreeqq, 2022-07-18:
after add own APN settings "available on device cdc-wdm0 because profile is not compatible with device (modem is incompatible with connection: device has differing sim-operator-id than GSM profile)"

```
Jul 18 20:27:46 tici ModemManager[4750]: <debug> [1658168866.915185] [modem0/ttyUSB2/at] --> 'AT+QENG="servingcell"<CR>'
Jul 18 20:27:46 tici ModemManager[4750]: <debug> [1658168866.944385] [modem0/ttyUSB2/at] <-- '<CR><LF>+QENG: "servingcell","NOCONN","LTE","FDD",260,01,A1EF06,392,3526,8,2,2,A029,-110,-14,-80,10,17<CR><LF><CR><LF>OK<CR><LF>'
Jul 18 20:27:46 tici ModemManager[4750]: <debug> [1658168866.946031] [modem0/ttyUSB2/at] device open count is 2 (close)
Jul 18 20:27:57 tici ModemManager[4750]: <debug> [1658168877.073282] [modem0/ttyUSB2/at] device open count is 3 (open)
Jul 18 20:27:57 tici ModemManager[4750]: <debug> [1658168877.073966] [modem0/ttyUSB2/at] --> 'AT+QTEMP<CR>'
Jul 18 20:27:57 tici ModemManager[4750]: <debug> [1658168877.197480] [modem0/ttyUSB2/at] <-- '<CR><LF>+QTEMP: 39,36,35<CR><LF>OK<CR><LF>'
Jul 18 20:27:57 tici ModemManager[4750]: <debug> [1658168877.198605] [modem0/ttyUSB2/at] device open count is 2 (close)
Jul 18 20:27:57 tici ModemManager[4750]: <debug> [1658168877.248549] [modem0/ttyUSB2/at] device open count is 3 (open)
Jul 18 20:27:57 tici ModemManager[4750]: <debug> [1658168877.248730] [modem0/ttyUSB2/at] --> 'AT+QNWINFO<CR>'
Jul 18 20:27:57 tici ModemManager[4750]: <debug> [1658168877.264214] [modem0/ttyUSB2/at] <-- '<CR><LF>+QNWINFO: "FDD LTE","26001","LTE BAND 8",3526<CR><LF><CR><LF>OK<CR><LF>'
Jul 18 20:27:57 tici ModemManager[4750]: <debug> [1658168877.264808] [modem0/ttyUSB2/at] device open count is 2 (close)
Jul 18 20:27:57 tici ModemManager[4750]: <debug> [1658168877.280592] [modem0/ttyUSB2/at] device open count is 3 (open)
Jul 18 20:27:57 tici ModemManager[4750]: <debug> [1658168877.280773] [modem0/ttyUSB2/at] --> 'AT+QENG="servingcell"<CR>'
Jul 18 20:27:57 tici ModemManager[4750]: <debug> [1658168877.310984] [modem0/ttyUSB2/at] <-- '<CR><LF>+QENG: "servingcell","NOCONN","LTE","FDD",260,01,A1EF06,392,3526,8,2,2,A029,-110,-13,-82,10,17<CR><LF><CR><LF>OK<CR><LF>'
Jul 18 20:27:57 tici ModemManager[4750]: <debug> [1658168877.311475] [modem0/ttyUSB2/at] device open count is 2 (close)
Jul 18 20:28:05 tici NetworkManager[3601]: <info>  [1658168885.4013] audit: op="connection-update" uuid="d3e5a78d-7286-4656-b53a-53b5fd372ea3" name="blue-prime" args="gsm.apn" pid=52027 uid=1000 result="success"
Jul 18 20:28:05 tici NetworkManager[3601]: <info>  [1658168885.4372] audit: op="connection-activate" uuid="d3e5a78d-7286-4656-b53a-53b5fd372ea3" name="blue-prime" pid=52027 uid=1000 result="fail" reason="Connection 'blue-prime' is not available on device cdc-wdm0 because profile is not compatible with device (modem is incompatible with connection: device has differing sim-operator-id than GSM profile)"
Jul 18 20:28:07 tici ModemManager[4750]: <debug> [1658168887.449329] [modem0/ttyUSB2/at] device open count is 3 (open)
Jul 18 20:28:07 tici ModemManager[4750]: <debug> [1658168887.449579] [modem0/ttyUSB2/at] --> 'AT+QTEMP<CR>'
Jul 18 20:28:07 tici ModemManager[4750]: <debug> [1658168887.572208] [modem0/ttyUSB2/at] <-- '<CR><LF>+QTEMP: 39,36,35<CR><LF>OK<CR><LF>'
Jul 18 20:28:07 tici ModemManager[4750]: <debug> [1658168887.573962] [modem0/ttyUSB2/at] device open count is 2 (close)
Jul 18 20:28:07 tici ModemManager[4750]: <debug> [1658168887.628259] [modem0/ttyUSB2/at] device open count is 3 (open)
Jul 18 20:28:07 tici ModemManager[4750]: <debug> [1658168887.628478] [modem0/ttyUSB2/at] --> 'AT+QNWINFO<CR>'
Jul 18 20:28:07 tici ModemManager[4750]: <debug> [1658168887.645776] [modem0/ttyUSB2/at] <-- '<CR><LF>+QNWINFO: "FDD LTE","26001","LTE BAND 8",3526<CR><LF><CR><LF>OK<CR><LF>'
Jul 18 20:28:07 tici ModemManager[4750]: <debug> [1658168887.647210] [modem0/ttyUSB2/at] device open count is 2 (close)
Jul 18 20:28:07 tici ModemManager[4750]: <debug> [1658168887.665391] [modem0/ttyUSB2/at] device open count is 3 (open)
Jul 18 20:28:07 tici ModemManager[4750]: <debug> [1658168887.666089] [modem0/ttyUSB2/at] --> 'AT+QENG="servingcell"<CR>'
```

pd0wm, 2022-07-27:
Pushed a fix for setting the APN: https://github.com/commaai/openpilot/commit/ab236e5ae0330edd984c8f19e694ab50090e9183

Does that fix your issue?

Piotrreeqq, 2022-07-28:
@pd0wm solved the problem :)

closed 2022-07-28 by Piotrreeqq