commaai/openpilot #25221: Kia Niro 2019 - "Check FCA" warning when engaging openpilot

DerGue69, 2022-07-20:
### Describe the bug

I have a Kia Niro EV 2019 (EU - Austria - 1st Edition). A "Hyundai H" Harness is my connector.
After installin stock OP my car was not recognized. I had to add values to values.py from my cars fingerprint (see additional info)
(I thought My Niro EV is already compatible with OpenPilot)

After that OP started working with my car. But everytime I activate OP I get a "Check FCA ( Forward Coll. Avoidance Assist)" warning. 

I recorded a route and made public.
https://cabana.comma.ai/?route=6111bd7df5043394%7C2022-07-20--14-47-58&exp=1689861829&sig=9eylZ8Csg1lvrMr6yneTzSZbSC7qN5ZjymwZ2jOTkVM%3D&max=-1&url=https%3A%2F%2Fchffrprivate.azureedge.net%2Fchffrprivate3%2Fv2%2F6111bd7df5043394%2Fc7e80e23a0c545d687808655907ee3ee_2022-07-20--14-47-58

Please help me to get rid of this error ;)

[image]

### Which car does this affect?

Kia Niro EV 2019 (EU - Austria - 1st Edition)

### Provide a route where the issue occurs

6111bd7df5043394%7C2022-07-20--14-47-58

### openpilot version

0.8.14

### Additional info

[image]

'carFingerprint': 'KIA NIRO EV 2020',
               'carFw': [{'address': 2004,
                          'ecu': 'eps',
                          'fwVersion': b'\xf1\x00DE  MDPS C 1.00 1.04 56310Q4100\x00 4DEEC104',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 2000,
                          'ecu': 'fwdRadar',
                          'fwVersion': b'\xf1\x00DEev SCC F-CUP      1.00 1.02 96400-Q4000         ',
                          'responseAddress': 0,
                          'subAddress': 0},
                         {'address': 1988,
                          'ecu': 'fwdCamera',
                          'fwVersion': b'\xf1\x00DEE MFC  AT EUR LHD 1.00 1.03 95740-Q4000 180821',
                          'responseAddress': 0,
                          'subAddress': 0}],

sshane, 2022-07-20:
Hi, can you try the branch on this PR? https://github.com/commaai/openpilot/pull/25223

Enter https://installer.comma.ai/commaai/niro-ev as the set up URL after uninstalling openpilot

sshane, 2022-07-20:
I see you removed your car from that list, was it causing you problems?

DerGue69, 2022-07-20:
I tried this both things from Discord Chat - But I had no success. (see below)
I will try  https://installer.comma.ai/commaai/niro-ev tomorrow and give you an update asap.

_sunnyhaibin — Yesterday at 9:31 PM
But remove CAR.KIA_NIRO_EV from this line: https://github.com/commaai/openpilot/blob/master/selfdrive/car/hyundai/hyundaican.py#L19
after that, reboot
Just to test
Steering is fine
Image
also worth removing from this: https://github.com/commaai/openpilot/blob/master/selfdrive/car/hyundai/carcontroller.py#L132_

sshane, 2022-07-20:
Btw do you also trying this branch after? https://github.com/commaai/openpilot/pull/25122

It's a fix for the VIN queries not returning correctly, and it happens your car is affected

DerGue69, 2022-07-21:
> 

Installed "niro-ev" branch:

- Car was recognized automatically. No need to add fingerprint.
- When engaging OP, again I get the message "FCA ....".
- I shared the route  >> 6111bd7df5043394%7C2022-07-21--09-36-37 << for troubleshooting.
[image]

https://cabana.comma.ai/?route=6111bd7df5043394%7C2022-07-21--09-36-37&exp=1689925644&sig=K2wVYIUymbPoaDHnAEK6AwvNuUcYMOIVJ6ihLzTSIHo%3D&max=1&url=https%3A%2F%2Fchffrprivate.azureedge.net%2Fchffrprivate3%2Fv2%2F6111bd7df5043394%2F05fad89595576b1e986e75ebb6623f03_2022-07-21--09-36-37

DerGue69, 2022-07-21:
I updated the vin.py like in https://github.com/commaai/openpilot/pull/25122 as requested.
I have the same issue. FCA warning.
Shared affected route at: 6111bd7df5043394%7C2022-07-21--10-27-05

https://cabana.comma.ai/?route=6111bd7df5043394%7C2022-07-21--10-27-05&exp=1689928255&sig=LsG%2FeJVfRaHna2668JbOSzOqGHUU4kWGVNt49%2FWbLNc%3D&max=-1&url=https%3A%2F%2Fchffrprivate.azureedge.net%2Fchffrprivate3%2Fv2%2F6111bd7df5043394%2F6f020ce2d536d1c8ccde85ebfd2a5b95_2022-07-21--10-27-05

sshane, 2022-07-21:
Can you try the branch on this PR? https://github.com/commaai/openpilot/pull/25237 Please make sure to actually check out the branch cleanly, not modify another branch

adeebshihadeh, 2022-08-13:
@DerGue69 have you had a chance to try the branch?

sshane, 2022-08-13:
Yes, we've been chatting on Discord. It works to hide the alerts, but his car is still sending abnormal signals around related to FCA and FCW. However he's still able to get FCW so I'm going to confirm with Greg how we should merge this.

DerGue69, 2022-08-13:
> @DerGue69 have you had a chance to try the branch?

Yes. Shane and I did lots of tests. This branch gets rid of the annoying warning FCA warning.
When driving too fast to a standing car, I still get a collision warning in the car.

sshane, 2023-08-08:
@DerGue69 have you tried removing/adding your car to the various lists in the same function we modify the fusion state?

DerGue69, 2023-08-28:
@sshane: What exactly do you mean with "removing/adding your car to the various lists"?

In openpilot/selfdrive/car/hyundai/hyundaican.py ...
in the section of if car_fingerprint in Kia Niro ... 
I added:

    # Some cars show an FCA fault when forwarding CF_Lkas_FusionState. Hide FCA Fault:
    values["CF_Lkas_FusionState"] = 0

[image]

I still have no idea, what the "CF_Lkas_FusionState" is - but it hides the FCA warning completely.
I cannot perform a real-world crash test - to check if the actual FCA is working. (lack of inflateable car)

Mirukuuu, 2023-11-14:
Hi, I just got my comma 3X and connected it to a Kia Niro EV 2019 in Europe and have the same warning as described. Has there been any further progress on this issue?

DerGue69, 2023-11-14:
Hey Mirukuuu, greetings from Vienna.
Shane and I tried lots of things. The last thing I did, I watched Cabana Code like in Matrix to find some clues.
I even made a crash test with my brother to get some data...

Long story short:
I made my own branch as a 99.99% clone from Master.
The 0,01 % difference is one line of code in : openpilot/tree/master/selfdrive/car/hyundai/hyundaican.py
**### values["CF_Lkas_FusionState"] = 0**

This makes the error vanish. The only thing "we" don't know, if the real FCA (FrontCollissionAssistent) is still working with that code added. (That's why I was planning a crash-test)

I use my 99.99% clone since 60k+ km without any issues and warnings:
https://github.com/DerGue69/openpilot/tree/master

Mirukuuu, 2023-11-15:
Thanks a lot DerGue69 for the quick reply and the extensive explanation! I'll see if I can try this out soon, seems easy enough.

DerGue69, 2023-11-16:
If you need help/tips ... let me know.

DerGue69, 2023-12-23:
> Thanks a lot DerGue69 for the quick reply and the extensive explanation! I'll see if I can try this out soon, seems easy enough.

Any updates on your integration?

DerGue69, 2024-02-13:
> Thanks a lot DerGue69 for the quick reply and the extensive explanation! I'll see if I can try this out soon, seems easy enough.

Any updates?