commaai/openpilot #25259: Hugging right and too many DM warnings

kinng, 2022-07-23:
### Describe the bug

My car has been always hugging right since day 1.  I was changing the CAMERA_OFFSET in lane_planner.py after every release.  I filed a bug report previously but was deferred.  Hopefully this issue can be resolved once and for all this time.  I have to say the hugging is getting less and less with every release so it's improving.

All turns were hugging which was expected in laneless mode.  But straight roads should be centered but is constantly hugging right.

DM logic seems to be too sensitive.  Sometimes when looking down the road in the front and still paying attention, DM would beep.  Turning the head right then left to look at the mirrors to check for car centering would sometimes trigger the warnings as well.

### Which car does this affect?

Toyota Highlander Hybrid XLE 2017

### Provide a route where the issue occurs

9ff6adaf72343a2c|2022-07-22--13-35-37

### openpilot version

0.8.15

### Additional info

Also check out route 9ff6adaf72343a2c|2022-07-22--14-30-24.

adeebshihadeh, 2022-07-24:
Can you connect your device to wifi so it can upload the logs?

kinng, 2022-07-24:
Should be done, according to **connect**.  If not, left me know what else to do.

YassineYousfi, 2022-07-25:
Could you point us to a segment or two (in the two routes you shared) where your car was hugging right on a straight road?

ZwX1616, 2022-07-25:
> Sometimes when looking down the road in the front and still paying attention, DM would beep.
Were those alerts near the beginning of the drive? DM does need to "calibrate" for your driving position every drive but it won't take long

> Turning the head right then left to look at the mirrors to check for car centering would sometimes trigger the warnings as well.
Warnings are expected when looking into side mirrors for an extended amount of time.

kinng, 2022-07-25:
> Could you point us to a segment or two (in the two routes you shared) where your car was hugging right on a straight road?

Strange, some segments from the 1st route are gone from connect.  Anyway, huggings are from the 2nd route from time 14:38:07 to 14:38:24; and from 14:50:45 to 14:52:48.  Note all turns hug too but I am assuming it's due laneless mode.

> Were those alerts near the beginning of the drive? DM does need to "calibrate" for your driving position every drive but it won't take long

Yes it seemed that way.  It's got more stable toward the end of drive.

> Warnings are expected when looking into side mirrors for an extended amount of time.

I think it's about 3 seconds of time.  I do that quite often in prior releases and seldom get a warning on that.

YassineYousfi, 2022-07-25:
It doesn't seem like there is any hugging in the 2nd route. The car position is at y=0. 
The figure is from 30 seconds in <14:50:45 to 14:52:48> where you weren't overriding the steering. The model seems to drive at max 10cm far from the center with no clear bias.
[image]

kinng, 2022-07-25:
I wonder if you can detect my manual overriding to the steering wheel.  Most times I have to override (slight steering to the left) on local streets.  14:50:45 to 14:52:48 is on local street.

YassineYousfi, 2022-07-25:
Yes, the small vertical gray line is one short steering override. I tried to find a stretch where you didn't constantly override, and in that stretch it didn't seem to hug.

kinng, 2022-07-25:
The beginning of the 1st route is more obvious, though it's the first turn-on on 8.15 so it might be calibrating at that time.  But that route and the 2nd route are gone from connect so I can't even review them anymore.  My determination of hugging is thru driving experience and the left right rear mirrors.  Not sure if your data can indicate that, otherwise it should have corrected the problem.

kinng, 2022-07-25:
Not sure if it's helpful, but in 8.14, I changed CAMERA_OFFSET from 0.04 to -0.05 to make is almost center.  I think probably -0.06 is better but it's already good enough.

YassineYousfi, 2022-07-26:
This shouldn't be necessary. The logs show that when you stopped overriding the steering, the model correctly positioned in the lane. Feel free to post if you have other segments with consistent hugging in a straight road. Preferably with less steering overrides.

kinng, 2022-07-26:
Ok, just took a quick trip to a straight road. Route is 9ff6adaf72343a2c|2022-07-25--21-05-00--2
No overriding this time.  Best segments are from 21:08:00 to 21:10:00.  Note the white car on the right lane at 21:09:26.  It was moving behind me in the right lane but for I think too afraid to pass because I was too close to it's lane.
The whole road is not exactly flat (going downhill from left to right) so that may have affected the centering logic.  In fact on the return trip and car initially hugged left and adjusted to center and eventually hugging right just a little, much less than the initial direction.
Since many cars don't have the hugging problem, I really suspect some data regarding my car is probably off, because the hugging has been very consistent since the beginning.

kinng, 2022-07-27:
I am pretty sure it's hugging right all along and I doubt the graph shows the real-life picture.   When you saw it got close to the right lane but in reality it crossed the right lane.   I think what the software saw as center in reality is not really center.  I think there is some offset somewhere that's not set right for my car.

ZwX1616, 2022-07-27:
> I think it's about 3 seconds of time. I do that quite often in prior releases and seldom get a warning on that.

This release has some big DM changes, so changes in alerting behavior are expected, such as the ones from your drives. I would say it is normal as long as they are not unfair.

kinng, 2022-07-28:
> This release has some big DM changes, so changes in alerting behavior are expected, such as the ones from your drives. I would say it is normal as long as they are not unfair.

This I can live with.  Afterall, I did move my head away from driving.  But a slight tilting the head down but still very much looking ahead and paying attention would trigger the warning is really too sensitive and will be very tiring if I have to hold the head high.

YassineYousfi, 2022-07-28:
>  I think what the software saw as center in reality is not really center. I think there is some offset somewhere that's not set right for my car.

The graph shows an accurate picture of where your device was relative to the lane lines, I see from your wide cam video that your device might be off center. Can you try remounting it to be better centered?

Please re-open this issue if it still hugs after that.

closed 2022-07-28 by YassineYousfi

kinng, 2022-07-28:
My mounting has been tried multiple times and was discussed previously.  See https://discord.com/channels/469524606043160576/524327905937850394/905896523806674994
 and 
https://discord.com/channels/469524606043160576/874396554222534677/906720171870269471

Currently the calibration is 0.05 up and 0.8 degrees left.  As far as center mounting is concerned, I am quite confident is very close to center already.  I have tried moving C3 about 4 inches to the LEFT and it didn't make much difference...i.e. still hug right.  Will try moving to the right and see what happens.

kinng, 2022-07-29:
Here's another picture of my mount, this time using a fishing line along with a laser light for alignment from the antenna fin to the Toyota logo and both confirmed my mount is quite centered.  

https://photos.app.goo.gl/RQHPZBAURdR7HQJJ9

Ignore the aftermarket dashcam right above C3.  The one above the dashcam and within the trapezoid housing is the car's front camera which is dead centered.

Another picture where I put a pole up aligned to the Toyota logo and you can see it's pretty squared: https://photos.app.goo.gl/HtRNSF5Lnm3gnLWe7

To test out what happens mounting C3 to the right (toward the passenger's side), I installed another mount about 9.5cm to the right from the old mount.  Chose 9.5cm because it's the same distance I adjusted via CAMERA_OFFSET.  Did another drive: 

9ff6adaf72343a2c|2022-07-28--20-02-31--2

and indeed it centered quite well, even though now the calibration is 6 degrees down and 2 degrees right.  See if this helps to isolate where the problem is.

I am now even suspecting my C3's front cameras may not even be seated properly.

I don't like the current off-center mounting position because it's blocking my passenger's view.  A software solution like changing CAMERA_OFFSET is better.