commaai/openpilot #25270: Runaway Ping ponging

ns3284, 2022-07-25:
### Describe the bug

Every so often, car gets into a ping ponging state where it bounces back and forth between lane edges. Sometimes it recovers, other times it runs away and requires manual stabilization. 

Severe case can be found starting at
4395faadf0a373f2|2022-07-24--22-38-04--42
And ending at
4395faadf0a373f2|2022-07-24--22-38-04--43

### Provide a route where the issue occurs

4395faadf0a373f2|2022-07-24--22-38-04--43

### openpilot version

0.8.16 Nightly (e259656997)

### Additional info

2022 Chrysler Pacifica Hybrid

adeebshihadeh, 2022-07-25:
Can you try this branch https://github.com/commaai/openpilot/pull/25118 that uses the [new torque controller](https://blog.comma.ai/0815release/#torque-controller)?

ns3284, 2022-07-26:
I'll switch branches tomorrow and give it a shot, but no plans on highway driving to get a real test.

adeebshihadeh, 2022-08-01:
Have you had a chance to try the branch?

ns3284, 2022-08-01:
Have not, sorry. I did see that nice new feature added to make changing
branches easier though. Won't have any long drives until this weekend.

ns3284, 2022-08-13:
I'll switch branches tomorrow and use it this weekend.

ns3284, 2022-08-15:
In process of switching now. Was away from home computer, and thought I could use the built in "Change Branch" feature. That feature did not work, so manually adding remote branch and changing now.

ns3284, 2022-08-29:
Sorry for long delay, hadn't had much opportunity to try out the branch.
Been using it the last couple weeks and had a longer drive yesterday. So
far it seems to address the oversteering/oscillating.

ns3284, 2022-08-29:
One thing to note about this branch. Every so often the device shows
unrecognized vehicle and I need to reboot to get it back to functioning.

ns3284, 2022-09-15:
Been using for some time now. The initial issue seems to be fixed on this branch.

sshane, 2023-09-12:
Any updates on this? Can we use torqued like we wanted to wait for? @nuwandavek