commaai/openpilot #25301: EV6 Wind (HDA 1) - no OP steering

hoomoose, 2022-07-28:
### Describe the bug

OP does not steer the Kia EV6 Wind trim. This trim level has HDA1 instead of HDA2. I have no errors on my C3, all the buttons work, and auto-resume works. The EV6 Wind trim does not have an LKAS (0x50) signal, does not have the 0x2a4 signal, and only has a LFA signal (0x12a) on bus 5. This can be seen in Cabana with ID 5:12a. 

So far, I have removed references to 0x2a4 from all files in OP and in the Panda file safety_hyundai_hda2.h. I have tried replacing 0x50 with 0x12a. I have also tried setting the tx bus for 0x12a to can bus 1. 

My changes can be seen on:
https://github.com/hoomoose/openpilot/tree/ev6  
https://github.com/hoomoose/panda

Here is a Cabana link:
https://cabana.comma.ai/?route=68d6a96e703c00c9%7C2022-07-26--15-21-53&exp=1690523009&sig=UCSU%2ByMuTLx0YccTMfhynli3nLAI%2Bcqkupf8scniz4w%3D&max=3&url=https%3A%2F%2Fchffrprivate.azureedge.net%2Fchffrprivate3%2Fv2%2F68d6a96e703c00c9%2F36a57a6e01153a13db8e4d9ee48c51e7_2022-07-26--15-21-53

### Which car does this affect?

Kia EV6 Wind 2022

### Provide a route where the issue occurs

68d6a96e703c00c9%7C2022-07-26--15-21-53

### openpilot version

0.8.16

sunnyhaibin, 2022-08-12:
Current Status:
- The author of this issue is replacing his comma three due to a potential hardware issue.

Previous Actions:
- Pin swapped on the harness connectors to flip the can bus pairs. This put the messages on either bus 4 or bus 6 after verification on cabana, instead of bus 5 with the previous configuration.
- Edited openpilot and panda safety code to accommodate the bus changes after the pin swap.
- The EV6 HDA1 does not have the `LKAS 0x50` for openpilot to intercept and send steering commands therefore `LFA 0x12a` is used.

Potential Root Cause:
- With the previous harness configuration, the required messages were on bus 5, which openpilot could not intercept and send steering commands.

adeebshihadeh, 2022-08-23:
Now that we have upstreamed a CAN-FD HDA1 car, we know that the HDA2 cars are architecturally pretty dfiferent. @hoomoose can you get a route with the CAN bus lines flipped on your harness? You can ask how to do it in the HKG channel on discord or email us and we can ship one out to you.

hoomoose, 2022-08-23:
Here is one of my last routes with the CAN bus lines flipped. However, I have not been able to do any testing since 8/5 as my C3 upon car starting shows up with a dim camera-view screen and the message of camera malfunction. I haven't heard anything from Support since Fri 8/12 when I was told support is still looking into the issue.
https://cabana.comma.ai/?route=68d6a96e703c00c9%7C2022-08-05--00-38-11&exp=1691217869&sig=nSRE%2F0z%2BhK053clIbLBZEj0DCPaPYAeui9btu0pW2LM%3D&max=1&url=https%3A%2F%2Fchffrprivate.azureedge.net%2Fchffrprivate3%2Fv2%2F68d6a96e703c00c9%2F0bef5c9037f35018207feb7d5c0164bc_2022-08-05--00-38-11

hoomoose, 2022-08-25:
I have no idea why, but today my c3 started working again with no camera issues. So, I started troubleshooting.

Here is what I have done:
https://github.com/commaai/openpilot/compare/master...hoomoose:openpilot:0ev
https://github.com/commaai/panda/compare/master...hoomoose:panda:0ev

I receive no errors on startup, but when I activate cruise control I am getting a check blind spot safety system warning on my dash and a Controls Mismatch on the C3.

Here is a Cabana:
https://cabana.comma.ai/?route=68d6a96e703c00c9%7C2022-08-24--23-25-55&exp=1692942476&sig=FVhBIwgkVWvShXoW6phxwWFWMRVCm9IJ58Ue%2BJFOC2c%3D&max=-1&url=https%3A%2F%2Fchffrprivate.azureedge.net%2Fchffrprivate3%2Fv2%2F68d6a96e703c00c9%2Facdc1d51e763d8025b7b9f130a819de0_2022-08-24--23-25-55

hoomoose, 2022-09-01:
SOLVED:
Hyundai L was required.
Panda bug was found and fixed.

closed 2022-09-01 by hoomoose