commaai/openpilot #25311: Subaru: EPS faults on 2020+ models at low speeds

adeebshihadeh, 2022-07-30:
EPS seems to temporarily fault relatively frequently below 35mph.

scullionw, 2022-07-31:
I see you marked this as fixed here https://github.com/commaai/openpilot/pull/25248. 

What ended up being the fix? Was it just really just lowering the torque rates? I could swear I tried that on my crosstrek..

adeebshihadeh, 2022-07-31:
> I see you marked this as fixed here #25248.
> 
> What ended up being the fix? Was it just really just lowering the torque rates? I could swear I tried that on my crosstrek..

Oh, I meant that it's being moved to this issue. Though lowering the max torque and rates did help significantly, which we wanted to do anyway. Can you post some segments (ideally rlogs uploaded and preserved) where your Crosstrek faults?

scullionw, 2022-07-31:
ah gotcha!

I just tried lowering both rates all the way down to 17 and torque to from 1439 to 700 in case i had somehow not reduced it enough last time I tried. Unfortunately it still faults..

I'll try to post a segment soon.

scullionw, 2022-07-31:
22f2f49bcbecd829|2022-07-31--15-39-52 

- there are 4 warnings in this one, some of which i touched the wheel
- the one @ 15:42:09 i didn't touch the wheel

scullionw, 2022-07-31:
an older route with same issue (faults at 130, 252, 282 in the following cabana): 

https://cabana.comma.ai/?route=5cbf0beff26ee650%7C2022-02-25--01-19-02&exp=1677293732&sig=5y%2FjWb6AWDGP3UZy2gtj05clKfUuXSVMSirusl7Y0tw%3D&max=6&url=https%3A%2F%2Fchffrprivate.azureedge.net%2Fchffrprivate3%2Fv2%2F5cbf0beff26ee650%2Fccaca8f1f12415f3bd64a7c9e74856d4_2022-02-25--01-19-02

closed 2023-09-07 by jnewb1