commaai/openpilot #25343: No Panda, AEB/ACC disabled Chrysler Pacifica 2022

ns3284, 2022-08-02:
### Describe the bug

Git Commit 4f0fe747ba on master-ci today. Tried rebooting, tried restarting car, etc. Comma keeps showing "No Panda" and car states on start "Auto Emergency Braking (AEB) / ACC Limited Functionality Service Required" though I don't see a check engine light.

### Which car does this affect?

Chrysler Pacifica 2022

### Provide a route where the issue occurs

N/A as route won't generate, Serial: 4395faadf0a373f2

### openpilot version

0.8.16 Nightly (4f0fe747ba)

adeebshihadeh, 2022-08-02:
https://github.com/commaai/panda/pull/1006 fixes the "No Panda", but your cruise fault is an unrelated issue. Does the fault occur with the harness installed **without the comma three plugged in** (make sure to unplug the device while the car is off, so the faults can clear)? If so, sounds like you may have a bad harness.

ns3284, 2022-08-02:
As mentioned, the cruise fault may be some odd symptom. My CEL does not come on, like it normally would when there is a fault. I'll wait for that Panda fix and report back, or we can close this as a duplicate issue.

adeebshihadeh, 2022-08-02:
As I said, these should be unrelated issues, and I'd like to explore the fault. If they are not unrelated, then that is its own issue that should be addressed. Either way, there's a problem here that we should understand.

ns3284, 2022-08-02:
[image]
[image]
[image]
Just some visuals of what I saw.

ns3284, 2022-08-03:
Latest update has fixed the problem. I don't see the error on the car anymore. I wonder if somehow the signal from the sensor where the harness is connected may have had some corruption which caused that warning. Basically, perhaps this will always happen when Panda is unable to load?

Achilles308, 2022-08-03:
> Latest update has fixed the problem. I don't see the error on the car anymore. I wonder if somehow the signal from the sensor where the harness is connected may have had some corruption which caused that warning. Basically, perhaps this will always happen when Panda is unable to load?

I also got a brief cruise fault when I had the no panda error on master-ci recently. My car cleared it after the relay clicked over though... the error was only on for maybe 1/10 to 1/4 of a second? Maybe your car kept it for longer?

closed 2023-04-19 by adeebshihadeh