commaai/openpilot #25347: Openpilot’s steering is very jerky while taking bends (Hyundai Kona EV 2022)

Zack1010OP, 2022-08-03:
### Describe the bug

Even at Low speeds my Kona EV 2022 takes the bend with very jerky steering. It does take it properly but the jerkiness is probably related to a tuning issue as shared by Shane 

also sometimes when it is taking a round about to get off an express way, it hugs the side too much instead of being centered in the lane (you can also see this in the bend in my video)

you can see in the video about 5:21pm I had to intervene otherwise I would have crashed into the side. 

At about 5:12pm you can see an example of the jerky steering wheel

### Provide a route where the issue occurs

ff973b941a69366f|2022-08-03--17-09-29--0

### openpilot version

0.8.16

adeebshihadeh, 2022-08-13:
Can you try this branch? Uninstall and re-install with `installer.comma.ai/commaai/kona-torque-tuning`

Zack1010OP, 2022-08-13:
Yes sir will do so by today!

Warmest regards,

Isaac

adeebshihadeh, 2022-08-16:
@Zack1010OP plots look good on the new tune; how does it feel compared to the previous tune?

Zack1010OP, 2022-08-16:
Hi, the turning is smoother now but it keeps over turning

1) ff973b941a69366f|2022-08-15--11-25-26--0
11:29am you can see it turned so much that i almost careened into the inside barrier

2) ff973b941a69366f|2022-08-15--22-45-08--0
11:55pm it turned so hard to the left that it changed lanes by itself

3) ff973b941a69366f|2022-08-16--12-39-15--149
3:09pm a gently curving road but it didn't keep to the centre of the lane properly and almost crashed into the car on the left 

Additional question - on the Comma Connect's timeline when i'm pulling up the clips for you, what does it mean when the timeline is blue vis a vis green?

adeebshihadeh, 2022-08-16:
> Hi, the turning is smoother now but it keeps over turning

This is likely not a tuning issue, but the model wanting to cut the turn. The high torque of your Kona just happens to expose this particular model failure. We're working on the turn cutting and hope to ship improvements in 0.8.16.

> 1. ff973b941a69366f|2022-08-15--11-25-26--0
>    11:29am you can see it turned so much that i almost careened into the inside barrier
> 2. ff973b941a69366f|2022-08-15--22-45-08--0
>    11:55pm it turned [...]

Can you send connect links of these specific issues? Then, I can confirm it's the model and not tuning. You can zoom in on the timeline to the specific event and just copy the link to that zoomed in segment from the address bar.

> Additional question - on the Comma Connect's timeline when i'm pulling up the clips for you, what does it mean when the timeline is blue vis a vis green?

It's the same meaning as the border color on the device: blue is disengage and green is engaged.

PR #25426 (2022-08-16): Hyundai: Kona torque tune
closes #25347
closed 2022-08-16 by adeebshihadeh