commaai/openpilot #25349: 2017 Jeep Grand Cherokee - Dashcam Mode

wangstein, 2022-08-03:
### Describe the bug

My comma is stuck in Dashcam mode for days. I am trying to follow the fingerprinting instructions but the drives that are uploaded all says "mock" so i'm unsure how to get the proper carparams.

My car is actually a Jeep Grand Cherokee 2017, but the successful uploads says it's Grand Cherokee 2018. I guess which is fine.

### Which car does this affect?

Jeep Grand Cherokee 2017

### Provide a route where the issue occurs

c88f65eeaee4003a|2022-08-02--22-53-59

### openpilot version

0.8.15

sshane, 2022-08-04:
This started happening after you updated to 0.8.15, so it's likely the longer fingerprint time gathering more fingerprints than what we have added for your car. This will be fixed soon with fingerprinting 2.0 with firmware versions, but for now you can follow the fingerprinting guide here: https://github.com/commaai/openpilot/wiki/Fingerprinting#fingerprinting-10

Let me know when you have a fingerprint to add!

wangstein, 2022-08-04:
Should i also follow instructions for Fingerprinting 2.0 as well? There's a great video > https://www.youtube.com/watch?v=foIsRG0kMAo but I noticed that both all say on a successful drive/upload. Does it matter that I pick a route that the platform is mock for value? or do i have to pick a drive that still had the right car type in there?

wangstein, 2022-08-04:
@sshane > @jyoung8607 helped me with this. His links are as follows:

The technical details are contained in two PRs:
https://github.com/commaai/openpilot/pull/25362
https://github.com/commaai/openpilot/pull/25363

closed 2022-08-04 by wangstein