commaai/openpilot #25369: 0.8.16 master-ci oversteering on roads curving towards the right

ynwng, 2022-08-05:
### Describe the bug

Branch: 0.8.16 master-ci
Car; Toyota Corolla Hatchback 2020

0.8.16 seems to oversteer when the road curves right.

From the route:
- 19.16.02 - Felt like the car was going to run into the curb.
- 19:18:11 - Car cutting into the shoulder of the road.
- 19.19.07 - I also felt like the car was going to run into the curb. There was a warning to take over.
- 19:19:18 - Car driving into the next lane on the right.

I'm not sure if cutting into the next lane/shoulder is the intended behavior of laneless since I was previously using Shane's fork on 0.8.13 without laneless. To me, switching to master-ci feels less chill and more stressful, since I feel like I can't trust openpilot as much.

### Provide a route where the issue occurs

50c35b8d29fd7574|2022-08-04--19-05-32

### openpilot version

0.8.16

ynwng, 2022-08-05:
Some more on 50c35b8d29fd7574|2022-08-04--18-20-49

- 18:26:44 - Starts driving into the right lane
- 18:29:34 - Drifting off into the shoulder, but maybe an acceptable amount? Did not feel comfortable but laneless is new to me.
- 18:30:01 - On an offramp, I thought it was going to turn into the curb.

ynwng, 2022-08-05:
Not sure if I should create a separate issue but I noticed this model driving over the left lane line sometimes too.

50c35b8d29fd7574|2022-08-04--12-20-06--0

- 12:21:49 - Engaged openpilot and it started driving over the left line. The wheel was past the double yellow lines when looking through the driver side mirror (12:21:52). When it started getting close to the median and the pole, I tugged the wheel to the right just to be safe I wouldn't hit it, but didn't disengage (12:21:58).
- 12:22:53 - Driving over the double yellow line to the left. You can see me tug the wheel back to the right at 12:22:55.

ynwng, 2022-08-07:
Some more segments before I switch back to Shane's fork on 0.8.13. All of the timestamps below were handled fine by 0.8.13 though.

- 50c35b8d29fd7574|2022-08-06--18-28-00--2 (18:30:41) - Driving on the lane line for a right bend, which was not ideal for the car next to me which slowed down thinking I was trying to change lanes and taking awhile to do so.
- 50c35b8d29fd7574|2022-08-06--18-53-45--5 (18:59:34) - Already have a different clip of this, but 0.8.16 cannot stay drives in the middle of two lanes for this bend.
- 50c35b8d29fd7574|2022-08-06--19-45-28--1 (19:47:01) - Would have hit the car parked on the shoulder if I didn't disengage.
- 50c35b8d29fd7574|2022-08-06--19-45-28--4 (19:49:41) - Not the issue of the right lane but driving on the double yellow does not feel safe.
- 50c35b8d29fd7574|2022-08-06--20-38-02--1 (20:39:21) - Would have hit the cones but I tugged the wheel left.
- 50c35b8d29fd7574|2022-08-06--20-38-02--5 (20:42:55) - Went way out in the shoulder on a right bend, as if the shoulder was part of the lane.

HaraldSchafer, 2022-08-08:
The model does oversteer in tighter turns sometimes. We are looking into this and trying to fix it. Thanks for the detailed feedback!!

HaraldSchafer, 2022-08-12:
https://github.com/commaai/openpilot/pull/25413

Hopefully this improves your experience

ynwng, 2022-08-19:
Hey Harald, I've been running the falcon punch model for a few days now and the right oversteering is much better. It will still sometimes drive over the right lane on right bends, but less harshly and I guess that's to be expected out of laneless. This issue seems closeable on my end. I did run into a new bug, but I'll create a separate issue for that.

ynwng, 2022-08-19:
The separate issue that I created https://github.com/commaai/openpilot/issues/25487

closed 2022-08-19 by HaraldSchafer