commaai/openpilot #25382: Toyota: `NO_STOP_TIMER_CAR` hybrids consume power during standstill

cydia2020, 2022-08-07:
### Describe the bug

It has come to my attention thanks to a Discord user's tip-off that Toyota `NO_STOP_TIMER_CAR` hybrids may unnecessarily waste power when the vehicle is at a standstill with openpilot longitudinal engaged. The user reported the car uses around 400 watts of power when stopped with openpilot engaged, but only wastes around 100 watts with the stock system under the same conditions.

I believe this is due to openpilot's implementation of the `NO_STOP_TIMER_CAR` logic, where some hybrid cars won't go into standstill when *openpilot longitudinal is stopping behind a lead vehicle* (this issue does not apply when the user is manually stopping behind a lead car and enabling openpilot after they have came to a complete stop).

I have been procrastinating to report this issue due to lack of access to proper equipment, my Chinese clone of the XHorse cable was unstable at best, it recently kicked the bucket, so I have decided to purchase an OBDLink LX. With that piece of equipment in hand, I have been able to personally verify this with my MY2018 Prius.

I will try to conduct some tests on this issue, and document my findings here. Please, if you have access to the proper hardware, test this on your vehicle as well, the more data the better.

### Which car does this affect?

Presumably all toyota hybrids marked as no stop timer car

### Provide a route where the issue occurs

NA

### openpilot version

all

cydia2020, 2022-08-07:
Test 1:

Testing procedure:

Please note, this test may be unreliable, but it gives us some insights as to what is going on. I will conduct more tests in the coming days with better logging.

Data logged:

Motor (MG2) Current (A)
Motor (MG2) RPM
HV Battery Voltage (V)
HV Battery Current (A)

The test:

I engaged openpilot with the vehicle stopped shortly after logging started on the OBDLink app, I waited a few seconds for the graphs to stabilise, and looked at the data recorded by the OBDLink app.

I then slightly stepped on the accelerator pedal, enough to get the vehicle moving, but not sufficient for the speed sensor to register, I observed the data again.

Findings:

[image]
Graph 1 - the vehicle is in standstill, ready to be resumed with a tap of the accelerator or the resume button, around 1 amp of current is flowing from the HV battery to the rest of the vehicle. The HV Battery’s voltage sits at roughly 220V, and is stable.

[image]
Graph 2 - I tapped on the accelerator, as can be seen by the spike in the out-flowing current, and current rushing into the motor. The vehicle moves forward slightly, but not enough for the speed sensors to register, openpilot does not recognise this as a standstill condition, so my vehicle goes into “standstill hack” mode. The HV Battery’s voltage starts to drop.

[image]
Graph 3 - The vehicle is in “standstill hack” mode for a while, around 3.2 amps of current is flowing from the HV battery to the rest of the vehicle, the HV battery packs voltage continues to drop.

Here is a screen recording of this: https://www.youtube.com/watch?v=y790_-sHlRo

cydia2020, 2022-08-08:
# Ideal test: 

## Prerequisites:
1. A TSS-P or TSS2.0 (or Lexus equivalent) "Hybrid Synergy Drive" vehicle that is capable of Full Speed Range Dynamic RADAR Cruise Control.
2. An OBD-II dongle that supports Toyota’s proprietary PIDs.
3. Configure your OBD-II device to read and log the following data
4. Hybrid battery voltage
5. Hybrid battery current
6. Vehicle speed
7. A relatively level intersection
8. The ability to enable/disable the "stop and go" hack in openpilot.

## How to enable/disable the "stop and go hack":

### To enable (if your car is capable of full speed drcc but is not in the `NO_STOP_TIMER_CAR` list):

Comment out these two lines in `selfdrive/car/toyota/carcontroller.py`:
```python
    if CS.out.standstill and not self.last_standstill and self.CP.carFingerprint not in NO_STOP_TIMER_CAR:
      self.standstill_req = True
```

### To disable (if your car is already in the `NO_STOP_TIMER_CAR` list):
Remove `and self.CP.carFingerprint not in NO_STOP_TIMER_CAR` from L67 of `selfdrive/car/toyota/carcontroller.py`

## Test 1:

### Testing condition:

With full stop and go enabled, the car does not go into standstill upon reaching a stop, and is able to automatically resume once the lead car starts moving without any user intervention.

### Testing procedures:

1. Go somewhere near your intersection of choice.
2. start logging on your OBD-II software.
3. locate a suitable vehicle to follow.
4. engage openpilot and let openpilot follow the vehicle to a complete stop without any user intervention.
5. let openpilot automatically resume.
6. Drive for 10 seconds.
7. end data logging.
8. Save logged data as a CSV file, and upload it to GitHub 

## Test 2:

### Testing condition:

With full stop and go disable, the car enters standstill mode upon reaching a full stop, and requires user intervention (resume button click/accelerator depression) to resume.

### Testing procedures:

1. Go somewhere near your intersection of choice.
2. start logging on your OBD-II software.
3. locate a suitable vehicle to follow.
4. engage openpilot and let openpilot follow the vehicle to a complete stop without any user intervention.
5. Resume openpilot after the lead begins to move.
6. Drive for 10 seconds.
7. end data logging.
8. Save logged data as a CSV file, and upload it to GitHub.

cydia2020, 2022-09-18:
Closing, #25024 partially resolves this.

closed 2022-09-18 by cydia2020

mattszcz, 2022-12-08:
@cydia2020 can we reopen this issue? 

I can confirm this is still an issue in 0.9.1 in my '21 RAV4 Hybrid. When OP brings my vehicle to a stop, the yellow arrows remain illuminated in the power diagram indicating the electric motor is consuming battery power trying to accelerate.

Fixing this bug will reduce brake pad wear and increase MPG in stop and go scenarios on who knows how many vehicles. Not just Prius.

ErichMoraga, 2022-12-09:
> @cydia2020 can we reopen this issue? 
> 
> 
> 
> I can confirm this is still an issue in 0.9.1 in my '21 RAV4 Hybrid. When OP brings my vehicle to a stop, the yellow arrows remain illuminated in the power diagram indicating the electric motor is consuming battery power trying to accelerate. And it does this whether they stop and go hack is enabled or not.
> 
> 
> 
> Fixing this bug will reduce brake pad wear and increase MPG in stop and go scenarios on who knows how many vehicles. Not just Prius.

Any brake "pad wear" you think you have from that is negligible.  I agree the power consumption aspect is still worth some attention... though easily avoided if you disengage OP.