commaai/openpilot #25386: openpilot audio alerts stopped working

tecandrew, 2022-08-08:
### Describe the bug

Toggling openpilot did not result in any audio indication on the comma device of it enabling/disabling. Though lateral control did work; I can see cruise control enabled and the green border on the UI of it working. A reboot fixed the issue.

### Provide a route where the issue occurs

eb5a50f9e98f50b9|2022-08-06--18-55-53

### openpilot version

0.8.15-release3

### Additional info

A side affect of this bug: I used the intermittent "Cruise Control Conditions Not Met" alert on my Hyundai dashboard as an indicator that openpilot was disengaged and I appreciated this 😄 

Maybe always use this alert on the CAN bus on every disengage or as a fallback mechanism when audio fails?

adeebshihadeh, 2022-08-08:
> eb5a50f9e98f50b9|2022-08-06--18-55-53

Did sounds not work for the whole route or stop mid-drive?

> Maybe always use this alert on the CAN bus on every disengage or as a fallback mechanism when audio fails?

I don't think this alert really makes sense for disengagements. If I remember correctly, the stock system only uses it like openpilot's `no_entry` alerts that block engagement attempts.

tecandrew, 2022-08-08:
the whole route it didn't make a sound. I rebooted at the end of the drive and it resumed behavior as normal.

adeebshihadeh, 2022-08-13:
Moving to #22681. This should happen very rarely; let us know if it's happening frequently.

closed 2022-08-13 by adeebshihadeh

tecandrew, 2022-08-14:
noted. will keep a lookout