commaai/openpilot #25387: Failed to Lane Change

tecandrew, 2022-08-08:
### Describe the bug

A turn signal lane change failed when it I attempt to lane change from a freeway onramp. 

It then attempted to change lanes and took too long for comfort to complete the lane change. It was driving on top of a lane a little too long...

### Provide a route where the issue occurs

eb5a50f9e98f50b9|2022-08-05--22-45-14--26

### openpilot version

0.8.15-release3

### Additional info

I remember this occurring a few times before, and thought it was a fluke, but this was a very noticeable one.

YassineYousfi, 2022-09-07:
Openpilot was not engaged in eb5a50f9e98f50b9|2022-08-05--22-45-14--26
Are you sure this is the correct segment?

tecandrew, 2022-09-08:
It might have been on the segment before or after, but I did not preserve the entire route to see it. I checked connect.comma.ai before submitting this issue.