commaai/openpilot #25389: Learner issue on Ram HD trucks

realfast, 2022-08-08:
### Describe the bug

I have been running a C3 on a Ram HD trucks since 10/2021. A large issue I have noticed is that the learner has a very hard time with these trucks. For example, the steering ratio is 15.61. However, I have seen the learner error from the liveParameters.steerRatio getting too high. As a test, I have raised the initial steer ratio to 25 and it errored once liveParameters.steerRatio crossed 50. Also, liveParameters.stiffnessFactor will continually learn down on a long trip. I have seen it get to 0.09 before if I disable the error. I am not positive about the exact cause of this. On several long turns, I have received the liveParameters.angleOffsetDeg error crossing over 10 as well. Attached is a log of the steering angle continuing to climb. The steering angle in the fork is set to 15.61. 

### Which car does this affect?

Ram HD

### Provide a route where the issue occurs

https://useradmin.comma.ai/?onebox=8fb5eabf914632ae%7C2022-08-07--11-32-11

### openpilot version

0.8.16

adeebshihadeh, 2022-08-08:
Some more Ram 2500 + 3500 dongle IDs
```
3f86423299323072
6038cd2b8addb4dc
8fb5eabf914632ae
b226af344c3cedd7
e98b5d809abaf860
ebf413d369473f7a
```

rob3210, 2024-06-09:
following

rob3210, 2024-06-09:
following

dparring, 2025-05-07:
Updating here because I have been working on this issue and have it partially fixed.  

The stock OP tune is mostly undrivable because the starting values for the learner are so far off.  Stiffness and steer ratio eventually run away uncontrollably and lead to boundary issues and paramsd errors as noted above.  With better starting points, the control loop finds a stable place and it is drivable.  I've confirmed this in logs and user feedback from about 5 trucks so far

I modified the stock tune and have it stable after making two key adjustments:  
- setting steer ratio to ~31
- adjusting actuator delay to 0.4  

I also updated the torque controller to use the steering angle sensor like most other OP cars.  The learners were stable on the original setting of use_steering_angle=False, however I found that oscillations were significantly dampened by setting it to True.

[image]

The only remaining issue is that the learner for steer ratio consistently converges on 2x of the actual steer ratio on every test truck.  The tune is driving well on multiple trucks, so it may only be a superficial issue in the code.  However I have not been able to narrow down the root cause of the doubling.  

I have confirmed that the truck self-reports a 15:1 steering ratio in its dash display based on visual steering wheel angle and front wheel angle.  And I have also validated that the angle sensor is mapped correctly in DBC and visually matches the steering wheel position.  I'm planning to look at STEER_STEP as a possible cause next.

Once this is fixed, the revised tune should be ready to upstream and will significantly improve the drivability on this platform.  There is also a related PR to fix the CC button mappings (https://github.com/commaai/opendbc/pull/2007) pending review.  Together, these might be enough to get this platform out of dashcam mode.

jyoung8607, 2025-05-08:
The steer ratio and tire stiffness learning process are closely linked and affect each other. As a guess, it's possible the learner is converging on wrong answers for both that happen to produce the right outcomes, for at least some driving conditions. 

If the starting tire stiffness is left at default, it'll be scaled by the configured mass [compared to a Civic](https://github.com/commaai/opendbc/blob/c856a2c0bd2b3c75f86a73b051c0c4cc7159559e/opendbc/car/__init__.py#L55). I wonder if that scaling breaks down when transitioning out of P tires and into LT tires, as one might when stepping up from a Ram 1500 (which works?) to a 2500HD or 3500HD (not so much).

dparring, 2025-05-08:
It is possible I found a combination that happens to work for the wrong reasons.  Most of the testing was performed with the learners disabled though, so I know that hard coding 31 ratio with 1.0 stiffness yields a stable drive.  I only turned the learners back on after the tune was stable with hard coded lateral accel, friction, ratio, delay, stiffness, etc.

Aside from the strange ratio, all of the other values are within normal ranges compared to other cars.  Since the DBC angle scaling appears to be fine, I'm working my way through the other variables in the learner equation, specifically the ones that come directly from the truck.  I don't think yaw rate or lateral accel come from it directly, so I'm focused on the torque feedback part of the loop now.  

It could be the tire component of that equation like you suggest.  The only reason I don't think that big truck tires are the root cause is because 100% of the 2500/3500 HD trucks I've looked at have learners that seek to 2x their real ratios, and none of the 1500s do.  And we have 1500s with wild tire combinations like 35's and 37's that drive ok.

Is STEER_STEP a dead end?  My theory is that a mismatch in frequency - say EPS processing at 25hz vs 50hz -  could lead to a 2x scaling factor appearing in the learned ratio.

jyoung8607, 2025-05-08:
> It could be the tire component of that equation like you suggest. The only reason I don't think that big truck tires are the root cause is because 100% of the 2500/3500 HD trucks I've looked at have [...]

What I'm thinking of isn't necessarily size, but [P vs LT construction](https://www.discounttire.com/learn/passenger-vs-light-truck). I'm not saying this is for-sure the problem, but it's a potential difference between the 1500 and 2500/3500s.

> Is STEER_STEP a dead end? My theory is that a mismatch in frequency - say EPS processing at 25hz vs 50hz - could lead to a 2x scaling factor appearing in the learned ratio.

Yeah that's a dead end. The params learner operates whether openpilot is engaged or not, so this won't be a problem with actuator control.

I would see what the params learner does if you set the initial steering ratio close to stock and make a large change to tire stiffness. I'm actually not sure which direction to take it, LT tires will be less grippy with deeper treads, but have stiffer sidewalls and higher inflation, so I'm not sure if the total slip angle will end up being worse or better than P tires. Try setting the tire stiffness factor to 0.5 and steer ratio to 15 and see what the learner does, then experiment again with stiffness factor 2.0. Not sure if this will help you, but it's the next thing I'd try.

dparring, 2025-05-08:
Great ideas, thank you.  Love the idea with locking stiffness at a factor of two and observing, will run that test next.  Will update here as I go, there is some public interest in this one

dparring, 2025-05-08:
Just documenting here as I go for tracking purposes.  Currently working backwards through the learner equations.  Base SR comes from here:

The road wheel angle 𝛿 is related to the steering wheel angle 𝜃 (measured by STEERING_ANGLE) via the steer ratio:

𝛿 = 𝜃 / steerRatio

​Angle comes from STEERING_ANGLE in the STEERING message in DBC.  The resolution is half degrees and the current mapping to angle degrees matches observed visual steering wheel angle (e.g. 90 degree sensor reading corresponds to 90 degree steering wheel turn).  Some HDs have angle display in the dash, so we were able to verify the 15:1 ratio there as well (e.g. 90 degree steering wheel angle displays a 6 degree wheel angle in dash).

Working backwards into the learner equation yields more avenues to investigate:

[image]

Vehicle speed (_v_) checks out ok.  Wheelbase (_L_) is reasonable and static.  Incorrect angleOffset would likely manifest as a directional bias and that has not been observed.  Next tests will focus on experimenting with stiffnessFactor as suggested by jyoung to see if there is a solution which leads to a stable control loop with a proper steer ratio.  Counter point to investigate on this is why 1500s are not manifesting in the same way, even though they are running on the same LT tires generally.

Note to self:  if this avenue doesn't pan out, investigate if angle sensor resolution could be playing a part (re: ZSS and Erich's work in Toyota).

dparring, 2025-05-13:
Progress update.  Testing tire stiffness is inconclusive on HDs:
- set initial ratio to 15x, stiffness locked at 0.5 = learner goal seeks to high ratio values, severe oscillation/overcorrecting
- set initial ratio to 15x, stiffness locked at 2.0 = learner goal seeks to high ratio values, severe oscillation/errors

Also observed a route with a non-TRX, non-Rebel 1500 on LT tires.  In contrast to HDs, ratio was relatively stable over a long drive and learned gradually down from the default.  So this leans against LT/P being the only factor.

I researched the Ram platform differences and it is possible that the discrepancy is coming from the solid front axle of the 2500/3500 compared to the IFS of 1500s (and every other OP car), specifically with how it affects lateral accel and road roll calculations.  It's the most significant known difference that fits the data so far and is consistent across the platforms.  I'm going to examine the road roll calculations and see if I can isolate that as the source of the discrepancy.

We have over 5k miles logged on the 30x tune so far and it's driving great for people, so we know a stable drive is possible for HDs.

15x ratio with 0.5 stiffness:
[image]

15x ratio with 2.0 stiffness:
[image]

Ram 1500 Limited with LT tires fitted:
[image]

jyoung8607, 2025-05-19:
Maybe rotational inertia? That's another thing that [won't scale like a Civic](https://github.com/commaai/opendbc/blob/447381dee2d30fb46e4bf1aec0529667d0b9d159/opendbc/car/interfaces.py#L162). The 2500/3500 trucks will have heavier suspension and other stuff distributed away from the CG, whereas the passenger cabin right over the CG probably isn't very different. Rotational inertia [figures into the vehicle dynamics model](https://github.com/commaai/opendbc/blob/447381dee2d30fb46e4bf1aec0529667d0b9d159/opendbc/car/vehicle_model.py#L187).

Strictly guessing (and anthropomorphizing) but maybe openpilot isn't properly feeling how heavy the truck handles and is constantly having to turn a little more, and thus confusing the learner?

dparring, 2025-05-20:
I suspect that the anomaly is either coming from where you suggest or from the Kalman code to derive lateral acceleration and roll from the device sensors.  There are magic numbers in both pathways that were empirically derived from observing IFS behavior.  I haven't been able to isolate which path or variable is the source of the discrepancy yet.  If the magic numbers are the source of the amplified steer ratio, then it's likely possible to "move" the discrepancy to other tuning parameters (e.g. wheelbase or mass), but only by also inflating them to fantastical values.

Is it possible to eventually get HDs moved out of dashcam mode with the high steer ratio, or is that going to be a blocker even if the tune is demonstrated to be stable?  @sunnyhaibin 's [PR](https://github.com/commaai/opendbc/pull/2007) for the 2024 buttons plus this tune gets the 2500/3500 platform reasonably on par with the 1500s in terms of user experience.  

I'm still prepping a PR but it will basically be this:  https://github.com/commaai/opendbc/compare/master...dparring:opendbc:master-ramhd-fixes  There is a bugfix for steering angle measurement included also.

As an aside, OP has massive tolerance for variance in the actual weight and steering dynamics of a vehicle.  This is the truck of one of our testers, who has 5,000+ stable miles on the 31x tune, both with and without the 17,000 lbs trailer: 

[image]

We have others driving on this tune with and without towing, and OP is unexpectedly resilient switching between the two - even on the same drive!

realfast, 2025-05-21:
The fix I always implemented was to lock the steering ratio in the learner. Locking this has given me almost 4 years of stable driving. If we know the exact steering ratio, I do not see the point of letting the learner modify this. For cars we do not know the exact steering ratio, I see it as beneficial.

dparring, 2025-05-21:
You were on the right track when you posted this.  The reason it errored out at 50 when you set to 25 was because there is a 2x sanity check in the learner.  When I started working on this, I locked the ratio but let the learner continue to wander so I could observe where it went in the logs.  Then I manually walked it up to 31 and found that the learner stabilized at that level.  Then I reenabled the learner and it continued to show a stable solution.  User feedback on overall drivability improved the closer I got to 30 manually, and stayed good after I turned the learner back on.  

I do think there is value in keeping the learner because I see a lot of variability between different Ram trucks in terms of where they eventually stabilize.  I'm currently using 31.5 as the happy value that starts everyone off at an average spot and lets the learner do its work for that specific truck.  Really the key fix for this issue is starting somewhat close to where the learner wants to go, otherwise it runs into boundary issues, runaway stiffness, and paramsd errors.

jyoung8607, 2025-05-21:
> Is it possible to eventually get HDs moved out of dashcam mode with the high steer ratio, or is that going to be a blocker even if the tune is demonstrated to be stable? [...]

If this really is stable and usable, I'm open to removing the dashcam flag, but the wacky steering ratio behavior creates a (rebuttable) presumption of something broken, which I think needs to be *understood* even if not fixed.

I'm glad that users are getting value out of openpilot on these platforms, but I worry that what's ""working"" today could be brittle, that it could break down when confronted with model updates or other changes, and then the current mystery could become a release blocker.

I'd like to look at some logs (not screenshots) to see if I can help make sense of it. Can you share some route IDs with uploaded logs from your testing? I'd like to see the truck do some normal straight-ish highway driving plus take some curves near its torque maximum at the 30-ish learned ratio and also fixed at the factory documented ratio, which @realfast seems to think also works. Right now I can't conceive of how both those configurations could drive equally well, so I'd like to compare data from both to see what I'm missing.

dparring, 2025-05-21:
Thanks for considering an update to upstream, and I totally agree that the ratio needs to be adequately explained and quality standards met.  I didn't want to put more time into upstreaming if the odd ratio was going to be a hard blocker.  These trucks have low resolution steering sensors and low torque, so it's possible they will never be good enough.  But I think it is worth a shot if we can get this stable for people.  This is part of my goal to get more high mount data into the firehose data set.

I'll prepare a set of routes based on your criteria to make review easy.  @jyoung8607, what branch do you want the routes on?  I've been testing on OP 0.9.7 with a patch for the HD buttons, but I can port the changes over to release or master and get routes there if preferred.

jyoung8607, 2025-05-21:
The best data would be from master (or release) with exactly the minimum changes needed to make it drive: remove the dashcam flag, set the initial ratio to 30, and anything else that might be necessary.

It would be helpful if you could also arrange a test route from a 1500 on the exact same code. The ratio change won't apply, of course, but it would be nice to compare with an otherwise similar vehicle that doesn't display the problem. 

I'd also like to make sure this data comes from non-dualies (different rear track width really changes vehicle dynamics) using stock or reasonably close to stock tires, without lifts or spacers or other suspension geometry changes.

dparring, 2025-05-21:
I think we have trucks that fit the bill, I'll start working on this.  Will be a few days at least and will update here as I go.  Appreciate the feedback!

jyoung8607, 2025-05-22:
@dparring In addition to the tests I already requested, you can try changing the [observation noise parameter](https://github.com/commaai/openpilot/blob/fdfba3f9f7792df5725260023d18f93bd503b094/selfdrive/locationd/models/car_kf.py#L74) to accommodate your low-precision steering angle sensor. The current setting of 0.05 degrees is appropriate for a sensor with 0.1 degree resolution, so for your sensor with 0.5 degrees resolution set it to 0.25 degrees. That way the filter isn't overly trusting of the inaccurate readings.

Make sure to nuke /data/params/d/LiveParametersV2 between all your tests, so that learning starts over.

@realfast I looked at an HD truck log, and there's some activity in 0x126 that looks like IMU output, probably lateral and longitudinal accel and a yaw rate. If you happen to have knowledge of the precise scaling and offsets, a PR to opendbc would be welcomed. This would give us another ground truth for use in verifying locationd and paramsd performance.

realfast, 2025-05-23:
@jyoung8607 I will send you some info.

dparring, 2025-05-30:
Progress update 5/30

I've been able to determine that some or all of the inflated steering ratio is coming from improper scaling of tire stiffness values in scale_tire_stiffness().  I'm working on re-tuning based on this finding, but the results are promising - we are currently iterating on a tune where the parameters are in normal ranges for OP cars, including SR (about 19x in the current iteration). 

The default front/rear stiffness values for HD trucks on stock OP are 434,366/539,506 N/rad, compared to the reference Civic values of 192,150/202,500 N/rad.  While there are certain aspects of the *tire* that will have higher stiffness than a Civic, this value as implemented in OP is a multivariate blended value of the tire's ability to effectuate lateral force when turned, meaning it also encapsulates current road conditions, tire pressure, steering geometry (including solid axle), tire vertical and lateral stiffness, and other factors.  I don't know enough about the dynamics model to comment on the naming or usage of this parameter.  However, I will note that both empirically in the field and logically in the code, the tire stiffness *as used in OP* for HD trucks should be significantly *lower* than a Civic.

There is also a related problem with tireStiffnessFactor that prevents its use to tune these values properly.  It is used both as a constant to scale the reference Civic values, and also as a dimensionless parameter live learned based on observed conditions.  This dual use causes paramsd errors because the learner is built around the assumption that the factor starts at 1.0 (it is reset explicitly each drive) and scales by a maximum of 5x up or down.  However if a vehicle sets a custom tireStiffnessFactor to scale the Civic constants, the learner learns from that value instead of rebasing the factor's range to 1.0.

Practically this means that the lowest functional (before throwing paramsd errors) tire stiffness value in current OP for HD trucks is tireStiffnessFactor=0.2, which scales to 86,873/107,901 N/rad, which just happens to be in the observed range that these trucks are stable in.  I'm re-tuning based on these findings with hardcoded values, if that yields a stable solution then I'll propose a backwards compatible facility for fixing the ability to tune tire stiffness without breaking the learner.

dparring, 2025-06-02:
Quick update here to document progress since I'll be away for a little and will pick this up when I am back.  The essence of the fix for this paramsd issue will look something like this:

First, decouple the learner from Civic constant scaling in scale_tire_stiffness() so that it rebases at 1.0 on the scaled values:
[image]

This unlocks the ability to scale the Civic tire stiffness constants without affecting the learner:
[image]

Tuning note: tireStiffnessFactor scaling as implemented is heavily dependent on vehicle mass, specifically once a vehicle's mass gets close to or above ~3000kg, the "allowable" minimum stiffness scales upward until you get to the point that the minimum value is higher than many of the cars in OP (as measured in N/rad, _not_ the scale factor).  A cursory look at opendbc says that Rivian is likely the next closest car potentially affected by this issue.  This also means you have to set mass and (to a lesser extent) center of gravity before tuning tireStiffnessFactor.

The result is an error-free drive and stable learners (from factory reset in this example):
[image]

I want to observe more routes after sequential drives and on more trucks, so I'll be working on the tuning values when I resume.  Then I'll prep the reference drives on master that @jyoung8607 requested and sort out how to approach the code changes.  The route above is on dongle ID 600021c572514cfb and code is dparring/RamReleasev0.9.7-jeffny-test if anyone with access wants to take a closer look.  I don't want to post the full route here since this tester shares routes to me publicly instead of by dongle ID.  Note that I haven't tried tuning on the observation noise parameter suggested above yet.

dparring, 2025-06-06:
Here is the same truck after about a week of driving and a few hundred miles without resetting anything:
[image]

I think I undershot the tireStiffnessFactor on this iteration (0.14) but overall the learners have stabilized and are giving stable results over long drives and multiple sessions.  User reported zero errors and is happy with the drive.

I'm going to split this fix up into two separate PRs, one to address the decoupling and one for the Ram HD tune itself.  I researched Rivian and discovered that they are running large stiffness values also: 366,031 N/rad front, and 578,621 N/rad rear.  The rear value appears to be the highest in OP due to the 50/50 center of gravity vs. the Civic.  I didn't find anyone reporting paramsd errors as a result of this, but they are currently struggling with tire differences between cars which may be related.

dparring, 2025-07-07:
Progress update 7/7

I updated https://github.com/commaai/opendbc/pull/2378 to allow front and rear tire stiffness to be set directly instead of using the scaling factor.  This unlocks the ability to tune them independently for cars like Rivian and is perfectly backwards compatible with existing platforms.  Using this capability, Ram HD trucks have a stable tune with "normal" values:

> `ChryslerCarSpecs(mass=3405., wheelbase=3.79, steerRatio=19, minSteerSpeed=16., tireStiffnessFront=65155, tireStiffnessRear=80926),`

There are multiple trucks driving on this now with good results.  I'll prepare test routes on stock master OP and submit a separate PR for the tuning update next.