commaai/openpilot #25437: niro ev(2020) dashcam mode Car Unrecognized

1gong2gong, 2022-08-15:
### Describe the bug

niro ev(2020), comma 3 , openpilot 0.8.15 installed but openpilot doesn't work.  dashcam mode Car Unrecognized message is displayed.  (Currently using neokii/op4 branch)

### Provide a route where the issue occurs

dong id : 9b052af2986bbe3c

### openpilot version

0.8.15

sshane, 2022-08-16:
Can you try the branch on this PR? https://github.com/commaai/openpilot/pull/25446

During setup for custom software, enter: https://installer.comma.ai/commaai/niro-ev-fp

1gong2gong, 2022-08-16:
> Can you try the branch on this PR and post a route? https://github.com/commaai/openpilot/pull/25446
> 
> 
> 
> During setup for custom software, enter: https://installer.comma.ai/commaai/niro-ev-fp

 Thank you.  
The vehicle was recognized well and the open pilot was executed.  

However, two problems occurred. 
1. When I run Openpilot, the message "Warning - this branch is not tested" pops up, but disappears when I tap the screen

2. If the comma3 reboots (device->reboot) while the engine is running, the FCW warning message appears on the instrument panel and disappears immediately.

sshane, 2022-08-16:
Good to hear, can you connect your device to WiFi to upload the route?

closed 2022-08-17 by sshane

1gong2gong, 2022-08-17:
> Good to hear, can you connect your device to WiFi to upload the route?

If i reboot from the device menu, the FCW warning message is appears and disappears immediately. (it does not appear while driving), 
but when i reboot after unplugging the USB C connector, the FCW warning message does not appear.

I still get the alert after doing a factory reset.  
I'm going to connect to my house so I can connect to Wi-Fi now.

To upload a route , do I activate record and driver camera in the device menu?

sshane, 2022-08-17:
You're rebooting your device with the car on? That's expected as openpilot stops sending messages the car is expecting. The relay flips a few seconds later and lets the car resume sending the LKAS messages