commaai/openpilot #25439: GM: set speed scaled by 1/16

twilsonco, 2022-08-15:
### Describe the bug

The dash set speed appears as 1/16th the value of the on-screen openpilot set speed.

Fixed by scaling up set speed before sending it to DBC, which was recently changed to handle the scaling dbc-side instead of in carstate.py.

[PR with fix submitted](https://github.com/commaai/openpilot/pull/25438)

### Which car does this affect?

GM

### Provide a route where the issue occurs

c11fcb510a549332|2022-08-15--11-11-45--0

### openpilot version

ee3d72a1e3be3247fcc2a853a29fdf951242b04d

twilsonco, 2022-08-15:
Here's from the posted route 
[image]

sshane, 2022-08-15:
Are you sure you updated your local submodules? You were also on a fork branch on this route

sshane, 2022-08-15:
You were on an old opendbc commit: `diff --git a/opendbc b/opendbc\nindex c6665ed1..48856851`

closed 2022-08-15 by sshane

twilsonco, 2022-08-15:
oops! Thanks @sshane