commaai/openpilot #25480: Wrong instructions at a roundabout when using navigation

DerGue69, 2022-08-18:
### Describe the bug

Instruction show first right, but it's 3rd right...

[image]

### Provide a route where the issue occurs

6111bd7df5043394|2022-08-18--17-58-07

### openpilot version

fix-lfa-warn

DerGue69, 2022-08-19:
OpenPilot v0.8.16

sshane, 2023-08-08:
Think this should solve this: https://github.com/commaai/openpilot/pull/29159

sshane, 2023-08-08:
I checked another route with a straight roundabout but mapbox gave us a "right" modifier which is incorrect, but it seems like we're parsing the data correctly. Nothing said straight in the route step.

Can you get a new route with rlogs uploaded?

DerGue69, 2023-08-08:
I would need a new prime abbo.

sshane, 2023-08-08:
That's fine, can you find the exact roundabout and send me a route with rlogs? I can simulate the rest

DerGue69, 2023-08-09:
@sshane This was last year on a road trip. 
This place is several hours away from home. 
So...no, sorry I can't.

I could find out the exact location on Google Maps with a little research.

sshane, 2023-08-09:
Oh that's fine, I'll use a recent route then you have and put the destination somwhere.

sshane, 2023-08-29:
@jakethesnake420 want to try this issue?

jakethesnake420, 2023-08-29:
@sshane sure I can take a look. Do i need to handle left hand drive? From skimming through the code I don't think navd is checking for 'driving_side'. Would also need to create flipped icons.

closed 2024-06-20 by adeebshihadeh