commaai/openpilot #25487: 0.8.16 master-ci changed lanes by itself

ynwng, 2022-08-19:
### Describe the bug

At 12:54:40ish, I engaged openpilot while I was behind a car. It moved to the right lane, and I think I saw openpilot still track it as the lead car while it was in the next lane (maybe around 12:54:48). This might have been because it was on a curvy road. Openpilot eventually stopped tracking it as the lead car. At around 12:55:01, openpilot starts driving over the right lane line - openpilot has always driven over this lane for me on 0.8.16 before and after the falcon punch model. Then openpilot changes into the next lane completely by itself, without me hitting the turn signal and nudging the wheel. I think maybe openpilot was drifting into the next lane and picked up the car in the next lane as a lead car, so it changed lanes to try to follow that car.

### Provide a route where the issue occurs

50c35b8d29fd7574|2022-08-18--12-48-45--6

### openpilot version

0.8.16

### Additional info

This is master-ci shortly after the falcon punch model got merged in.

ntegan1, 2022-08-22:
tssp corolla crosses lanes during low speed sharpish turns. many low speed lateral problems with this car

https://user-images.githubusercontent.com/3820605/186036630-69f7509c-2d96-45e8-ace0-d04f7b356647.mp4

a152dad3c3368320_2022-08-22--16-05-20--4

HaraldSchafer, 2022-08-25:
https://github.com/commaai/openpilot/pull/25524

Nuclear Grade model is far more accurate, hopefully it improves/fixes these issues @ntegan1 @ynwng

HaraldSchafer, 2022-08-26:
Going to close this since, the latest model should show improved behavior in these failures. Please feel free to reopen or make new issues if you experience new or similar issues. Thanks!

closed 2022-08-26 by HaraldSchafer

ynwng, 2022-08-29:
Thanks Harald, it works better now