commaai/openpilot #25492: Multilang: handle longer text in UI

sshane, 2022-08-19:
- [ ] Make toggle title text wrap like the descriptions<br>
  [image]
- [ ] (dynamically?) adjust button widths
  [image]

sshane, 2022-08-19:
@deanlee want to tackle any of these? The first should be a pretty easy one

bpluta, 2022-08-19:
@sshane I can try to do this if no one has taken this task yet. Dynamically adjustable text sizes would be nice feature as some phrases in different languages are much longer than in English and could not be made any shorter without statically shrinking text for all language versions

sshane, 2022-08-19:
Going to wait for a re-design instead

closed 2022-08-19 by sshane