commaai/openpilot #25493: sim: launch_openpilot.sh hangs at bootlog

jeroenlammersma, 2022-08-19:
### Describe the bug

Since commit [aa32ea0: loggerd: log all disk space usage in initData](https://github.com/commaai/openpilot/commit/aa32ea0f64e4035ac929e09ee85d268c3185221c#diff-b60345ac6d952fef3525d70194a17eebccfc84f5ce9c39d6537bfc707a5938e0) launch_openpilot.sh just hangst at (saving) 'bootlog to'. The UI doesn't show and openpilot doesn't start. See provided screenshot.

[image]

### OS Version

Ubuntu 20.04

### openpilot version or commit

03b074426a7de22b9f5dbfb9bb30f8325ac43b4a

equilibrium-liberalism, 2022-08-20:
similar issue reproduced in ubuntu 22.04 .

((openpilot) ) user:~/openpilot/tools/sim$     ./launch_openpilot.sh
**selfdrive/loggerd/bootlog.cc: bootlog to /home/hk/.comma/media/0/realdata/boot/2022-08-21--01-45-08**

but the replay by openpilot it self is fine [see](https://rewa-insights.com/t/awesome-automotive-autonomous/381/6?u=rewa-evija)

jeroenlammersma, 2022-08-24:
> similar issue reproduced in ubuntu 22.04 .
>
> ((openpilot) ) user:~/openpilot/tools/sim$ ./launch_openpilot.sh **selfdrive/loggerd/bootlog.cc: bootlog to [...]

Not sure what the exact issue was, because the page won't load anymore. But it seemed that you are using CARLA 0.9.12, which is outdated. Update to 0.9.13 and it will work again.

jeroenlammersma, 2022-08-24:
Updated requirements and it works again. Resolved.

jeroenlammersma, 2022-08-31:
Currently having the exact same issue again:

[image]

I am running master branch, updated requirements. Even did a full fresh and clean install on another machine and there it also keeps hanging on this step.