commaai/openpilot #25504: Cannot run can_print_changes.py: pandas package missing from AGNOS

JasonJShuler, 2022-08-20:
### Describe the bug

Unable to run can_print_changes.py due to missing dependency "pandas".

[image]

### Provide a route where the issue occurs

6b0ebba58cb3baba

### openpilot version

992ee329e96c80795bab2be98f979c0ae39ee5a2

### Additional info

I'm able to install the 'pandas' package on a per-boot basis using the following workaround:

```
sudo mount -o remount,rw /
sudo chmod 777 /usr/local/pyenv/shims
cd /data/openpilot
pip install pandas
```

[image]

adeebshihadeh, 2022-08-21:
We try to keep the AGNOS image size small, so we tend not to include large dev dependencies. However, we still want it to be easy to install what you want, e.g. AGNOS hooks `apt` to auto remount as rw. Here's the issue tracking doing the same thing with `pip`: https://github.com/commaai/agnos-builder/issues/65. For now, `sudo mount -o rw,remount / && sudo $(which pip) install pandas` will install pandas persistently (until the next AGNOS update).

closed 2022-08-21 by adeebshihadeh

JasonJShuler, 2022-08-29:
@adeebshihadeh that tool is incredibly useful - I wish I knew about it sooner, it could have saved hours and hours of staring at Cabana. Could I update the readme with the tip, or would it be possible to remove the Pandas dependency? It looks like it may have been added recently...

adeebshihadeh, 2022-08-29:
The `tabulate` package is in AGNOS. I'd definitely accept a PR to replace pandas with it.