commaai/openpilot #25525: 2022 CX5 OEM Front Camera Sensor Failure Error

kravatch, 2022-08-23:
### Describe the bug

When driving with OP active, front camera sensor error shows up on the OEM instrument cluster

similar error to https://vf.vin/en/car/dlrvc-mazda-cx5-2018-jm3kfbdm5j0335098.html 

Error goes away with disengage/rengage OP

No errors seen on C3 unit and self drive does not disengage and does not seem affected

### Which car does this affect?

2022 CX5 Signature

### Provide a route where the issue occurs

1bf23ce7dee4e6fc|2022-08-22--19-28-34

### openpilot version

0.8.15 release3

adeebshihadeh, 2022-08-23:
* Does this only show while engaged?
* Does it stay on once it appears?
* Does it appear instantly after engaging?
* How consistently does this happen? Every engagement?

@Jafaral do you know anything about this?

kravatch, 2022-08-23:
> * Does this only show while engaged?
yes
> * Does it stay on once it appears?
I believe so. I left it on for about 30 seconds after the warning triggered before I disengage OP which clears it
> * Does it appear instantly after engaging?
no, it does take some time, if the error happens (it does not always happen)
> * How consistently does this happen? Every engagement?
not every engagement, but it does correlate with more complicated actions (slow to stop and s-turning)

Jafaral, 2022-08-23:
Would help I cab view CAN data @adeebshihadeh

Jafaral, 2022-08-23:
@kravatch  are you running stock OP? or one of my branches?   if you are running stock OP, can you switch and try `mazda-0.8.15` ?

kravatch, 2022-08-23:
I am running stock 0.8.15

To clarify earlier behavior, the error does go away on its own, but does come back pretty consistently so long as self-drive was active

Unfortunately today my c3 unit started experiencing fan failures on the unit itself and it overheated and reset 2 times. I suspect that may be related to the camera failure. I am still within the 30 day purchase window and will be pursuing a hardware RMA

adeebshihadeh, 2022-08-23:
> Unfortunately today my c3 unit started experiencing fan failures on the unit itself and it overheated and reset 2 times. I suspect that may be related to the camera failure. I am still within the 30 day purchase window and will be pursuing a hardware RMA

The fan malfunction is software. We're pushing out 0.8.16 this week to address it. You can run the `release3-staging` branch to fix it for now. https://github.com/commaai/openpilot/pull/25509

kravatch, 2022-08-23:
@Jafaral I did just try the mazda-0.8.15 branch but the camera issue persisted and I also faced a "cannot steer under 0 mph" error (though it did self drive just fine)

I also needed to patch in my FW values IE https://github.com/commaai/openpilot/pull/25423

Jafaral, 2022-08-23:
@kravatch , can you please send me a drive with camera issue? You can dm me on discord.

kravatch, 2022-08-27:
I think i have narrowed the camera issue down to low(er) speed only

I suspect it may be an interaction with the OEM TJA in the signature trim which kicks in below 40 MPH

I have not been scientific in my experiments, but above that speed does seem to be around when I stop seeing this

Jafaral, 2022-08-30:
I did discover new signals in your car, working to see if can fix i the issue for you:

[image]

This also might be related to :  #24881 , to steer from zero without delay in cars with no stock TJA enabled.

sshane, 2023-01-17:
@kravatch going to close since you report this is no longer an issue on master. Please let me know if that's not the case!

closed 2023-01-17 by sshane

DouglasSherk, 2023-11-17:
I'm getting this same issue but on a 2024 Signature now. I'm on master-ci from yesterday.